zephyrproject-rtos/zephyr #5144: BUG: cmake: make doesn't update zephyr.hex file

zhaozhongchen, 2017-11-24:
zephyr doesn't update zephyr.hex file if it is already exist, even though zephyr.elf was updated.

SebastianBoe, 2017-11-24:
Could not reproduce.

Used sample ~/zephyr/samples/synchronization
and BOARD nrf52840_pca10056

on commit

96e7df466a928593e3d1c5ae25fc74709d71772b

Please post steps to reproduce.

SebastianBoe, 2017-11-24:
```
sebo@mach:~/zephyr/samples/synchronization$ md5sum b/zephyr/zephyr.hex 
7b25d4a8ae1d4cc5018e4dfdd23a93c0  b/zephyr/zephyr.hex
sebo@mach:~/zephyr/samples/synchronization$ make -C b
make: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[1]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[  2%] Built target offsets
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[  3%] Built target offsets_h
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
Scanning dependencies of target app
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[  4%] Built target linker_script
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[  5%] Built target linker_pass2_script
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
[  6%] Built target subsys__bluetooth__common
[  9%] Built target subsys__bluetooth__host
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[ 10%] Building C object CMakeFiles/app.dir/src/main.c.obj
[ 12%] Built target subsys__net
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
[ 28%] Built target subsys__bluetooth__controller
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[ 51%] Built target kernel
[ 83%] Built target zephyr
[ 84%] Linking C static library libapp.a
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[ 93%] Built target app
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
[ 94%] Linking C executable zephyr_prebuilt.elf
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[ 95%] Built target zephyr_prebuilt
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
[ 96%] Generating isr_tables.c
gen_isr_tables.py: (4189, 4625, 46, 0, 7)
gen_isr_tables.py: spurious handler: 0x105d
gen_isr_tables.py: Configured interrupt routing
gen_isr_tables.py: handler    irq flags param
gen_isr_tables.py: --------------------------
gen_isr_tables.py: 0x155d     0   0     0x0
gen_isr_tables.py: 0x1741     6   0     0x20003484
gen_isr_tables.py: 0x1a15     17  0     0x0
gen_isr_tables.py: 0x7e23     13  0     0x0
gen_isr_tables.py: 0x7e1d     24  0     0x0
gen_isr_tables.py: 0x7de9     11  0     0x0
gen_isr_tables.py: 0x7e27     1   1     0x0
Scanning dependencies of target kernel_elf
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make[2]: Entering directory '/home/sebo/zephyr/samples/synchronization/b'
[ 97%] Building C object zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj
[ 98%] Linking C executable zephyr.elf
Generating files from zephyr.elf for board: nrf52840_pca10056
make[2]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
[100%] Built target kernel_elf
make[1]: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
make: Leaving directory '/home/sebo/zephyr/samples/synchronization/b'
sebo@mach:~/zephyr/samples/synchronization$ md5sum b/zephyr/zephyr.hex 
e31821b5fb227f55ea0b3fe8ca0acb71  b/zephyr/zephyr.hex
sebo@mach:~/zephyr/samples/synchronization$ echo $BOARD
nrf52840_pca10056

```

SebastianBoe, 2017-11-24:
What might be happening, is that the platform you are on no longer defaults to generating hex files after commit 1f1143a , so you are seeing an old hex file built before 1f1143a.

You can test this by using a clean build directory.

zhaozhongchen, 2017-11-24:
yes, you are right.

somedays ago *make* create an old zephyr.hex file, and today i update my source code with master.

I remove all file in build and rebuild target, there are no file named zephyr.hex. It's my fault.

closed 2017-11-24 by zhaozhongchen