zephyrproject-rtos/zephyr #5145: samples/bluetooth: Connection failure on peripheral CSC with Arduino 101

tarunkum, 2017-11-24:
Peripheral Cycling Speed & Cadence sample app is designed to advertise CSC pkts. 
If any connect request comes from central device then this peripheral should connect successfully.
Once connection is establish then central device should be able to list the attributes of Peripheral device 
As well as Central device should be able to read the characteristics of Peripheral.
In zephyr version v1.9-branch, connection and reading the characteristics is getting successful , but with latest version  
connection is not occurring successfully once Central device try to get connect with peripheral then "connected" message doesn't comes on Zephyr console & connection failed on arduino_101.

 
 
 Tested zephyr version:
 Latest version: v1.10.0-rc1

 Logs
 ***** delaying boot 1000ms (per build configuration) *****
***** BOOTING ZEPHYR OS v1.10.0-rc1 (delayed boot 1000ms) *****
Bluetooth initialized
Advertising successfully started

How to reproduce.
1.Take one zephyr ble target(arduino_101) and ble dongle connected with Linux machine. And placed both BLE devices near to each other.
2.On zephyr target: Install peripheral CSC sample app($ZEPHYR_BASE/samples/bluetooth/peripheral_csc).
3.Check on console and waiting for "Advertising successfully started" on Zephyr console.
4.Run bluetoothctl tool on linux machine(where BLE dongle is connected).
5.Run below commands to connect BLE dongle to arduino_101.
 $ bluetoothctl
 $ power off
 $ power on
 $ remove * # remove all scanned devices from cache.
 $ scan on 	# scan for CSC advertisment pkt. waiting for CSC advertise pkt.
 $ scan off
 $ devices	#Look for CSC peripheral in list of devices & get the MAC address.
 $ connect $MAC_ADDR_A101 # $MAC_ADDR_A101 is a BLE mac addr of CSC peripheral.
 $ list-attributes	# Once the device is connected successfully then all the characteristics can be read one by one.

 
Expected result:
1.Zephyr target and ble dongle must be connected.

Actual result:
1.Zephyr target and ble dongle is not connecting.

tarunkum, 2017-11-24:
Same Connection issue seen in following sample apps as well as  :-
peripheral,
peripheral_dis,
peripheral_esp,
peripheral_hids,
peripheral_hr,
peripheral_sc_only.

jhedberg, 2017-11-24:
I just tried the latest Zephyr (both hci_uart and peripheral app) on my Arduino 101, and the connection was successful from BlueZ (bluetoothctl). So I cannot reproduce this issue.

tarunkum, 2017-11-28:
Thanks a lot Johan. I have upgraded the BLE firmware on Arduino 101 after that I did not face any issue related to connectivity.

closed 2017-11-28 by tarunkum