zephyrproject-rtos/zephyr #5148: Lightweight flash storage layer

carlescufi, 2017-11-24:
Zephyr already supports NFFS as a storage layer, but it might be a little bit too heavyweight for certain applications in memory-restricted ICs.

A lightweight flash storage layer similar to Mynewt's "FCB" would be useful for those applications:
https://github.com/apache/mynewt-core/tree/master/fs/fcb

lpereira, 2017-11-27:
How does it compare to [SPIFFS](https://github.com/pellepl/spiffs)?

carlescufi, 2017-11-27:
@lpereira I see that the author also has a project for an on-chip FS:

https://github.com/pellepl/niffs

nvlsianpu, 2017-11-28:
Regard SPIFFS - In techdoc (https://github.com/pellepl/spiffs/blob/master/docs/TECH_SPEC#L42) there is  noticed about that it using read-modify-write algorithm for writing to flash, what is probably unwanted for some devices we supported in zephyr.

nashif, 2017-11-28:
i think those filesystems are orthogonal to the feature being requested, which is a flash circular buffer, ie. very low level and not really a file system.

nvlsianpu, 2017-12-05:
As @nashif mentioned booth SPIFFS and niffs are to heavy components (especially spiffs). I see that suitable is to take FCB from mynewt.

FCB is basing on abstraction of sys/flash_map API and its flash_area memory description structure. I will keep this flash_area structure and replce flash_area_xxx calls by almost direct flash driver calls.
I don't see a gain in implementing something like flash_area/flash_map module for FCB as it always operate on a set of Flash pages of the same kind.
Any thoughts on taht?

carlescufi, 2017-12-06:
@nvlsianpu Regarding flash areas, can you take a look at Mynewt's "mfg" before you discard using them? This is also something we'd like to have in the future I believe: https://github.com/apache/mynewt-core/tree/master/sys/mfg

nvlsianpu, 2017-12-07:
My summary after discussion with @carles
As we consider adding capability similarly to these which mynewt MFG (manufacturing meta region) has it look like it is suitable to have module similar to mynewt flash_map abstraction in future. My idea is to port only necessary part of flash_area_xx  function (which are only flash access API for fcb) at this point.
I expect that this base module will be developing more in near future.

nvlsianpu, 2017-12-13:
This is development branch for this https://github.com/nvlsianpu/zephyr/tree/add_fcb

carlescufi, 2018-01-02:
A good candidate for an additional lightweight FS: https://github.com/geky/littlefs, with Mbed OS here: https://github.com/ARMmbed/mbed-os/tree/master/features/filesystem/littlefs

nvlsianpu, 2018-01-02:
RFC PR of FCB addition: https://github.com/zephyrproject-rtos/zephyr/pull/5537

nashif, 2018-01-16:
merged

closed 2018-01-16 by nashif