zephyrproject-rtos/zephyr #5149: Runtime non-volatile configuration system

carlescufi, 2017-11-24:
Many of the different subsystems in Zephyr might require storing and loading configuration data from non-volatile storage. To achieve a backend-neutral mechanism that allows for central loading and storing of the runtime configuration, a new subsystem is required that manages this centrally.

At the very least the following backends should be supported:
* FCB (see https://github.com/zephyrproject-rtos/zephyr/issues/5148)
* NFFS

Inspiration from the sys/config subsystem in Mynewt:
https://github.com/apache/mynewt-core/tree/master/sys/config

carlescufi, 2018-01-19:
>The module requires the singly-linked List abstraction as NFFS already has (SLIST from queue.h

Can you move it to `ext/utils/` or something similar?

carlescufi, 2018-01-19:
I'm thinking that it might be even possible to place it in `include/misc` instead of `ext/`. I think we have exceptions for non-Apache software that go into the codebase for specific cases like this. @nashif @galak can you comment on this?

nashif, 2018-04-03:
merged, closing

closed 2018-04-03 by nashif