zephyrproject-rtos/zephyr #5151: MPU flash region for storage area.

nvlsianpu, 2017-11-24:
Currently there is possible to allow write to whole FLASH or not allow writing to FLASH at all.
It should be possibile to have option which will allow writing to flash area  which is specific for data Storage. Thanks to that e.g. program flash will be stil preserve while e.g. file system will might operate propellery.

nvlsianpu, 2017-11-24:
cc @carlescufi

ioannisg, 2019-05-08:
Downside, is that this will eat an MPU region index. But it's doable. Is this something still needed @nvlsianpu ?

nvlsianpu, 2019-05-08:
So fare I didn't revived any request for this - this would be must have for critical devices which are not zephyr target today. Optionally for such protection we can use other SoC resource than MPU (on some targets).

For development purpose advantage of having that is help on identification of bugs caused by over/under-run  flash write., but situation is more complex as in the same time need to allow writes to DFU image-collection area/s.