zephyrproject-rtos/zephyr #5153: [RFC] Discussion of "cmake" vs "make" variables, aka "build environment" vs "work environment" setup

pfalcon, 2017-11-24:
There was some discussion of this issue between @SebastianBoe and me in https://github.com/zephyrproject-rtos/zephyr/issues/5140 and https://github.com/zephyrproject-rtos/zephyr/pull/5142 , and as those tickets are particular cases, and the discussion is far from complete, I'd like to create this ticket for wider and longer discussion.

So, idea I'd like to convey is that original Kbuild based systems consisted (at least) of the following, pretty distinct parts:

1. Build system per se, which compiled sources into binaries.
2. "Work with"/test/debug system, e.g. "make run", "make gdbserver", etc.

Some aspects of p.2 were controlled with make variables, e.g. one could say "make run QEMU_PTY=1" to run just this qemu invocation via pseudo-tty.

During CMake conversion, it seems that (almost?) all variables which controlled p.2 behavior were converted into CMake variables, which of course change workflow relying on those vars adversely, making there usage rather inconvenient.

The obvious solution is to convert them back into *make* variables, like they were originally, and how people expected to use them.

Two following counter-arguments were raised/can be imagined to be raised:

a) Due to @SebastianBoe: "But we want IDE integration..."
b) Was used on a different occasion by @mbolivar, so I guess could be raised here too: "But CMake can generate for Ninja and other build systems..."

My response to both is the same: Folks, when you want to use IDE, when you want to use Ninja, you want to use p.1 of the above, the "build system" part of Zephyr's control environment. p.2 is outside of your control, sorry. It innately and natively targeted at unix shell, make, and other command-line, specific tools. If you want to build p.2 for another tools, you'll likely need to work out a solution for that particular tool (a specific IDE, a specific make alternative). It might be that you could reuse some parts of unix command line p.2, but even you can't be sure for now, quiting @SebastianBoe https://github.com/zephyrproject-rtos/zephyr/pull/5142#discussion_r152958702 "I can't be more specific than that because we haven't yet figured out how this will work, that will come after CMake has stabilized and Windows support has improved."

In the meantime, command-line and make using people just want their productive workflow, which was disrupted by CMake migration, back. And my argument is that converting as much as possible p.2-related vars from CMake to make is a reasonable and viable approach.

This is of course just RFC, representing my partial and biased opinion. I'd like to invite all interested parties to share their outlook, so we all understand where we're heading, and in via which ways.

Thanks.

carlescufi, 2017-11-24:
@pfalcon I think that moving those variables to actual "runtime" (p2) with Make/Ninja should not preclude the option of later setting those via IDE, meaning that, unless I'm missing something I don't see why we cannot move those back to p2 for some reason that escapes me now. I agree that options like `QEMU_PTY` belong in Make and not CMake, since they have nothing to do with the actual building of the target but rather with the way the target is run?

SebastianBoe, 2017-11-24:
I suspect that instead of a migration of variables from CMake to Make, we can do both in many cases and keep everyone happy.

pfalcon, 2017-11-24:
@carlescufi 

> should not preclude the option of later setting those via IDE

That's somewhat ambiguous. My understanding is that folks like @SebastianBoe, @mbolivar treat CMake as an abstraction layer with its own closed-circuit semantics which can generate a spec for an abstract, arbitrary build-worker system.

Of course, that's not going to work if we add rules for a particular build-worker system, namely, make, because it would contain things like `qemu -serial $(if $(FOO),bar,baz)`, because only make can interpret that `$(if $(FOO),bar,baz)`.

My arguments:

1) That would be done only to "p.2" run/test/work part of the system.
2) p.2 part would be largely specific to any particular environment anyway, and variables for one "backend" won't make much sense for other. For example, how a GUI IDE user can appreciate the meaning of QEMU_PTY var?

pfalcon, 2017-11-24:
> I suspect that instead of a migration of variables from CMake to Make, we can do both in many cases and keep everyone happy.

I did in https://github.com/zephyrproject-rtos/zephyr/pull/5154 , but that's extra work, extra testing, extra maintenance. And for what? In most cases, those opts aren't useful outside of command line unix make env (IMHO).

carlescufi, 2017-11-24:
@pfalcon 

> Of course, that's not going to work if we add rules for a particular build-worker system, namely, make, because it would contain things like qemu -serial $(if $(FOO),bar,baz), because only make can interpret that $(if $(FOO),bar,baz).

Maybe I'm missing something here, but I believe one can generate code in CMake, using the CMake language, that then gets translated to the particular build system chosen. So @SebastianBoe, instead of writing `qemu -serial $(if $(FOO),bar,baz)`, can't we write something like `if(FOO in baz) qemu --serial` in CMake scripting language?

pfalcon, 2017-11-24:
> Maybe I'm missing something here, but I believe one can generate code in CMake, using the CMake language, that then gets translated to the particular build system chosen.

Maybe I'm missing something too. Per my understanding, CMake-based logic is static, in the sense that it happens once at cmake invocation. While we're talking about decision-making logic at make's invocation. I didn't hear about CMake language transplier to ~~JavaScript~~ make/shell language, but maybe I'm missing its advanceness indeed.

SebastianBoe, 2017-11-24:
> I did in #5154 , but that's extra work, extra testing, extra maintenance. And for what? In most cases, those opts aren't useful outside of command line unix make env (IMHO).

The benefit of, e.g. the technique of having flash_jlink, flash_openocd, targets instead of
a single target flash, which accepts make variables is that IDEs can know of the existance
of these targets and present them to the user.

As I said, we haven't started working on IDE support, so I don't have all the answers, 
but out of curiosity, I tried it out with CLion, and indeed, CLion presented every
custom target in a list of possible things that could "Run"; ram_report, flash, run, xconfig, etc.

So you can rest assured that it is not for nothing that we keep CMake aware of how a project
can be configured.

pfalcon, 2017-11-24:
> So you can rest assured that it is not for nothing that we keep CMake aware of how a project
can be configured.

Sure, I'm well aware that I have limited insight into this stuff, that's why I'm posted this RFC, to have a wider discussion not directly tied to a particular value.

> having flash_jlink, flash_openocd, targets instead of a single target flash

So, trying to translate that directly to "run" target, would "run_pty" work? It would, though "run" was  an abstraction over original "qemu" to allow running across various emulators, but running over PTY is a QEMU feature, though other emul's could support that too, why not? But with QEMU_PIPE that wouldn't work right away, because that var specifies a pipe path (not just a present-means-bool value).

Overall, it's clear that it's more complex structure than "build" vs "non-build" parts, some non-build parts (user-facing ones, like flashing) could take further evolution (instead of just bringing back to Kbuild), while more obscure like QEMU_PTY might be just brought back to older semantics. And some are "hard", like V=1.

nashif, 2017-11-27:
IMHO we are going back to how kbuild in zephyr was overloaded with things a build system is not supposed to be doing in the first place. Trying to make the **build environment** to act as a test-runner supporting different complex scenarios with different options, sometimes added adhoc does not scale and will make the build system very difficult to deal with, just like we had with kbuild. 
We do have most of the data we need in different places and available in the build directory in form of configs, cache etc. so could actually have dedicated test scripts and runners consume this data and run on a built kernel in a more efficient and manageable way with as many options as possible. This is valid for all the qemu related scenarios and complex networking setups but not limited to qemu and simulator invocation, the same can be applied on flashing, console handling and others. 

So lets not try and force cmake/make to do something they were not designed to do and try to step back and figure out a more sustainable and scalable solution.

(Hint: I have a POC already)

pfalcon, 2017-11-27:
@nashif : Some quick replies.

> different complex scenarios with different options, sometimes added adhoc does not scale and will make the build system very difficult to deal with, just like we had with kbuild.

I don't think that there was support for "complex scenarios", just the most basic ones, though I agree some were added in adhoc manner. And I'm not sure that Kbuild was hard to deal with because of these pretty modest additions, vs due to how it was built in the first place ;-). (The same applies to CMake too.)

> so could actually have dedicated test scripts and runners

Sounds good, but the matter, the usual direction is that projects have various adhoc scripts and tie them into the build system, not the vice-versa. It's very convenient to be able to do 90% of tasks with "make &lt;something>" vs scratching head trying to remember where lives that thing which does X. Take the following as an example: it's trivial to copy a built executable where user wants, and yet it's golden standard to provide "make install" (which works for a trivial case of one file and non-trivial case of directory trees of 1000s of files).

> (Hint: I have a POC already)

With the above in mind, looking forward to it! The current task is still to recover *some* important functionality which regressed with CMake conversion. ("Important" is for example related to regression testing.)

SebastianBoe, 2017-11-28:
I am sceptical to having a layer on top of Zephyr's build system.

The Zephyr build system is not just something that compiles code, nor should it be. It is a knowledge
engine that knows "how" to give a user something that the user knows "what" is. This doesn't mean
the knowledge engine always needs to know all the details, it is OK, for it to pass on a users request
verbatim to an underlying script, like a compiler driver does to it's underlying tools.

The problem with layering something on top of the knowledge engine is that the toplevel layer,
or script, is stupid compared to the lower layer, and you don't want your users interfacing with 
something that is stupid. You want them interfacing with something that knows as much as possible.

Any kind of upper layer would inevitably not understand what the user wants, and the user would then
have to change what it interfaces with and go to the layer that really knows what is going on.

This issue we are seeing with our build system technology not being flexible with it's user interface
does not justify introducing yet another tool that our users have to learn how and when to talk
to. There IMHO exist acceptable workarounds for it's CLI limitations, e.g. what was done in 
https://github.com/zephyrproject-rtos/zephyr/pull/5154/files#diff-e73668dbbcaf743dce6889d466f31bb0R24

That being said, I could be wrong, this is just speculation until the upper layer exists and we can see what the new user interface will look like for a user.

carlescufi, 2017-11-28:
@mbolivar FYI

carlescufi, 2017-11-28:
My current preference (which has changed a bit) based on these last days' conversations and personal usage of the build system is to have 2 layers:

* Expert users, integrators, test systems should always be able to, at the very minimum, build from CMake. Flashing is a nice extra, but I would personally not be against moving flashing entirely out of the build system and into an upper layer. Same goes for running. This is controversial since it might require a set of scripts for those expert users and it will move some of the functionality out of the CMake files and into a separate system, but if it's the only alternative to environment variables then I am in.

* Standard users can opt to use a (Python) tool that is transparent (i.e. simple to inspect and can easily print the commands required to flash and run) to build, flash and run. 

I am aware that this goes against the principle that we keep all building, flashing and running all part of  a single system. If we can find a reasonable way of doing everything with CMake (and that is compatible with both Make and Ninja) that is as user-friendly as passing command-line parameters to a tool that can "document itself" via a --help, then I will definitely take a look with interest, but it doesn't look like this is the case right now.

mbolivar, 2017-11-28:
@SebastianBoe wrote:
> The benefit of, e.g. the technique of having flash_jlink, flash_openocd, targets instead of
> a single target flash, which accepts make variables is that IDEs can know of the existance
> of these [...]

@pfalcon responded:
> So, trying to translate that directly to "run" target, would "run_pty" work? It would, though "run" was an abstraction over original "qemu" to allow running across various emulators, but running [...]

I believe @pfalcon has identified the crux of the matter here, which I have also had trouble dealing with. Specifically, it is sometimes necessary to pass values at runtime to targets in the generated build system.

Unfortunately, with CMake, the only portable way I can find to do this across generated build system types (and I'll out myself as a partisan against Make here; I think it's the past and ninja is awesome / the future) is with environment variables, which is undesirable as discussed at length elsewhere. If there is an other way I am missing, someone please speak up.

Here are some additional examples of runtime values from my experience:

- The unique identifier of a given board you want to flash, if several of the same type are attached (e.g. dfu-util's ``--serial``,  pyocd-flashtool's ``--board``, nrfjprog's ``--snr``). Support for this is a hard requirement if you want to do automated testing with real hardware.

- The name of an alternate binary file to flash. E.g., ``zephyr.bin`` is "good enough" if you want to flash a simple binary, but if you want to flash a binary that's been post-processed by mcuboot's tools to generate a signed blob, you need to be able to pass its path to the flasher.

In general, I think we should treat support easy flashing (brain-dead simple steps from "download zephyr" to "install blinky on aboard") as a hard requirement. There is a reason that Arduino is popular despite the fact that its firmware is terrible, and that's because it's dead easy to build and install programs. I think it's critical for Zephyr's adoption that we make this easy.

The latter example (flashing something other than zephyr.bin) is important to me because some information about how to flash mcuboot-loadable blobs is best obtained from the Zephyr build system itself (I'm talking specifically about ``CONFIG_FLASH_BASE_ADDRESS`` and ``CONFIG_FLASH_LOAD_OFFSET``). I have maintained mcuboot-aware build systems and applications targeting multiple boards for almost a year now, and let me tell you, you *really* don't want to duplicate this information.

@pfalcon originally wrote:
> p.2 is outside of your control, sorry. It innately and natively targeted at unix shell, make, and other command-line, specific tools.

I think this statement is making some unnecessary assumptions.

Specifically, apart from IDE use cases, there *are* command-line tools that aren't "natively targeted" at "unix shell" or "make", e.g. Python scripts. These have the benefit of also working on Windows without dragging along a bunch of Unix-specific tooling, which is why I've re-done the flash scripts in Python; Zephyr in general is moving in this direction, and I think it's smart.

@nashif wrote:
> So lets not try and force cmake/make to do something they were not designed to do and try to step back and figure out a more sustainable and scalable solution.
> 
> (Hint: I have a POC already)

Looks like several of us are orbiting around the same area, will be good to sync up on this further as soon as we can.

@carlescufi wrote:
> I am aware that this goes against the principle that we keep all building, flashing and running all part of a single system. If we can find a reasonable way of doing everything with CMake (and that [...]

This is a great summary IMHO.

SebastianBoe, 2017-12-31:
Did some experimenting with using environment variables for the "runtime" configuration.

It works kind of ok, if qemu was integrated into the runner infrastructure I think we would
 have something usable.

https://github.com/SebastianBoe/zephyr/commit/9973f2c10143e4c05db3c743784ec387cd9adbd1

carlescufi, 2018-01-19:
This is currently blocking https://github.com/zephyrproject-rtos/zephyr/issues/5756

carlescufi, 2018-01-19:
I will try to summarize the 3 options we have with regard to integration with external projects and going beyond the build:

1. Do nothing at all. Leave it to the users themselves, or to 3rd-parties, to document and/or manage the integration with external repositories, projects and tools.
2. Integrate the support into CMake with the help of additional support Python scripts.
3. Decouple completely the support for external repositories and tools and provide a Python-based tool that implements this.

The scope of what I have called "external repositories and tools" in my view would be:

* Bootloader integration. Including image signing and building the bootloader itself
* Advanced emulator support, including potentially networking setup with QEMU
* External repositories, including fetching and updating repositories that have known integration with Zephyr, such as MCUboot, OpenThread and others.

What I hope to achieve is to reach consensus to adopt one of the 3 options above and move in that direction from now onwards.

carlescufi, 2018-01-19:
@MaureenHelm and @galak FYI, in case you want to take part in the discussion.

mbolivar, 2018-01-20:
Those are also the options I see.

Option 1 is pretty bad usability for the core project, of course.

Lacking implementations for option 2 that don't involve environment variables for runtime arguments, I think it's a non-starter.

That leaves option 3, which I think can work with the proviso that "Decouple completely the support for external repositories and tools" means decoupling from CMake, not (necessarily) the Zephyr repository itself.

carlescufi, 2018-01-22:
>Lacking implementations for option 2 that don't involve environment variables for runtime arguments, I think it's a non-starter.

Since I believe that @SebastianBoe is a proponent of option 2, I would really like to hear the arguments here.

>That leaves option 3, which I think can work with the proviso that "Decouple completely the support for external repositories and tools" means decoupling from CMake, not (necessarily) the Zephyr repository itself.

Correct, that's how I meant it.

mbolivar, 2018-01-29:
>> Lacking implementations for option 2 that don't involve environment variables for runtime arguments, I think it's a non-starter.
> 
> Since I believe that @SebastianBoe is a proponent of option 2, I would really like to hear the arguments here.

Not sure who you are asking for details from, but from my side:

- the command lines would have serious usability problems
- the commands would need to be documented separately for Unix and Windows users
- sensitivity to environment variables makes diagnosing problems harder

On Bash:

```
VAR1="runtime-value1" VAR2="runtime-value2" ninja <target>
```

Users would need to learn this somewhat advanced shell feature, which is obscurely documented (https://www.gnu.org/software/bash/manual/html_node/Environment.html#Environment). They could be confused about why `ninja VAR1="runtime-value1" ... <target>` doesn't work (it does work with `make`!), and need extra help.

On Windows, this is the best I could find:

```
cmd /c "set "VAR1=runtime-value1" && set "VAR2=runtime-value2" && ninja <target>"
```

Source: https://superuser.com/a/223126 (and note from the replies that there are some subtleties here). I'm not sure what the equivalent would be on PowerShell.

By contrast, a wrapper program could be invoked with the same command line on all supported platforms, with a familiar and ergonomic syntax:

```
some_wrapper --opt1=runtime-value1 --opt2=runtime-value2 <command>
```

Environment variables also make diagnosing problems (and bug reports) harder, since if a user inadvertently leaves one set, then any command lines and output they share are missing critical information. Command line options do not have this drawback.

nashif, 2018-07-13:
Can we close this now?

carlescufi, 2018-07-13:
+1 for closing it