zephyrproject-rtos/zephyr #5177: hci_usb: Linking error

VinceGa, 2017-11-28:
GNU Arm Embedded Toolchain on OS X.
Version 6-2017-q2-update

[ 95%] Linking C executable zephyr_prebuilt.elf
libzephyr.a(usb_device.c.obj): In function `usb_data_to_host':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:198: undefined reference to `usb_dc_ep_write'
libzephyr.a(usb_device.c.obj): In function `usb_handle_std_device_req':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:557: undefined reference to `usb_dc_set_address'
libzephyr.a(usb_device.c.obj): In function `usb_set_configuration':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:457: undefined reference to `usb_dc_ep_configure'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:458: undefined reference to `usb_dc_ep_enable'
libzephyr.a(usb_device.c.obj): In function `usb_set_interface':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:513: undefined reference to `usb_dc_ep_configure'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:514: undefined reference to `usb_dc_ep_enable'
libzephyr.a(usb_device.c.obj): In function `usb_handle_std_endpoint_req':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:700: undefined reference to `usb_dc_ep_set_stall'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:681: undefined reference to `usb_dc_ep_is_stalled'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:690: undefined reference to `usb_dc_ep_clear_stall'
libzephyr.a(usb_device.c.obj): In function `usb_handle_control_transfer':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:225: undefined reference to `usb_dc_ep_read'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:228: undefined reference to `usb_dc_ep_set_stall'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:261: undefined reference to `usb_dc_ep_read'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:269: undefined reference to `usb_dc_ep_read'
libzephyr.a(usb_device.c.obj): In function `usb_enable':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:898: undefined reference to `usb_dc_set_status_callback'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:902: undefined reference to `usb_dc_attach'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:911: undefined reference to `usb_dc_ep_configure'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:916: undefined reference to `usb_dc_ep_configure'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:921: undefined reference to `usb_dc_ep_set_callback'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:925: undefined reference to `usb_dc_ep_set_callback'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:939: undefined reference to `usb_dc_ep_enable'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:943: undefined reference to `usb_dc_ep_enable'
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:932: undefined reference to `usb_dc_ep_set_callback'
libzephyr.a(usb_device.c.obj): In function `usb_write':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:976: undefined reference to `usb_dc_ep_write'
libzephyr.a(usb_device.c.obj): In function `usb_read':
/Users/VincentG/Zephyr/zephyr/subsys/usb/usb_device.c:982: undefined reference to `usb_dc_ep_read'
collect2: error: ld returned 1 exit status

It seems that the link order is not right. Those functions are declared in usb_dc.h
Most likely need an update in the make file.

finikorg, 2017-11-29:
How do you build it? I tested it works fine for quark_se_c1000_devboard board. What is your board?

SebastianBoe, 2017-11-30:
Unable to reproduce. 

@VinceGa : Please create a new issue with more detailed steps to reproduce and assign to me.

closed 2017-11-30 by SebastianBoe

VinceGa, 2017-12-07:
I tried to build hci_usb according to this instruction: https://devzone.nordicsemi.com/blogs/1059/nrf5x-support-within-the-zephyr-project-rtos/
The target board is NRF52840 PK. So I did "cmake -DBOARD=nrf52840_pca10056 .."

In the same environment, the other samples worked just fine. Only hci-usb gave me the linking errors. So I wonder if this has something to do with the  @CMAKE?

finikorg, 2017-12-07:
@VinceGa we do not have yet USB supported in Zephyr for this board.

VinceGa, 2017-12-07:
@finikorg what board does Zephyr support as a USB device then?
If you look at the error message, it has nothing to do with the board support.

VinceGa, 2017-12-07:
Might be related to https://github.com/zephyrproject-rtos/zephyr/issues/3798

finikorg, 2017-12-08:
https://github.com/zephyrproject-rtos/zephyr/pull/5320

finikorg, 2017-12-08:
@VinceGa supported USB drivers are: Designware USB Device Controller Driver and USB device controller driver for STM32 devices. See: drivers/usb/device/Kconfig

PR #5320 (2018-01-12): usb: Fix USB Device dependency
USB Device stack should depend on supported drivers not on menuconfig
option.

Fixes: #5177