zephyrproject-rtos/zephyr #5184: kernel system call handlers missing due to -Wl,--no-whole-archive

andrewboie, 2017-11-28:
Discovered by @agross-linaro, who was getting "unimplemented system call" errors on ARM for k_thread_abort().

ARM has a custom implementation of k_thread_abort under arch/arm. This has the effect of compiling out everything in kernel/thread_abort.c except the handler function.
For some strange reason, if --no-whole-archive is enabled, the linker decides to prefer the weak handler for k_thread_abort() in syscall_dispatch.c.

linkmeyer, 2017-11-29:
@andrewboie , can you please set the priority?

lpereira, 2017-11-29:
I set the priority to high since the way things are right now, they won't work on ARM.  This fix is required.

lpereira, 2017-11-29:
I had originally set this to high because this wouldn't work on ARM, but userland isn't working on ARM yet and won't be at least for 1.11. So moved back to medium.

andrewboie, 2017-11-29:
This issue affects x86 and needs to go into 1.10

andrewboie, 2017-12-13:
@AdithyaBaglody discovered that this problem is still happening, my patch didn't completely fix it.
Diving in....

andrewboie, 2017-12-14:
@AdithyaBaglody found a fix

commit 3ae64445 (2017-11-29): gen_syscall_header: create dummy handler refs

The core kernel is built with the --no-whole-archive linker option.
For all the individual .o files which make up the kernel, if there
are no external references to symbols within these object files,
everything in the object file is dropped.

This has a subtle interaction with system call handlers. If an object
file has system call handlers inside it, and nothing else in the
object file is referenced, then the linker will prefer the weak
version of the handler in the generated syscall_dispatch.c. The
user will get an "unimplemented system call" er [... cut]
closed 2017-12-14 by andrewboie