zephyrproject-rtos/zephyr #5189: tests/subsys/fs/nffs_fs_api:-Evalution failed

rrahulkx, 2017-11-29:
As on  zephyr commit ID:-07e1a64e77e15180721c56993a19bce473ff4365

**Steps to be Reproduce:**

cd $ZEPHYR_BASE/tests/subsys/fs/nffs_fs_api
mkdir build && cd build
cmake -DBOARD=qemu_x86 -DTEST=cache .. 
make run

**Then console output is obtained as pasted below:**

Running test suite nffs_fs_cache_test
===================================================================
starting test - test_cache_large_file

    Assertion failed at /tests/subsys/fs/nffs_fs_api/src/nffs_test_utils.c:216: nffs_test_util_assert_cache_range: cache_start not equal to expected_cache_start
FAIL - test_cache_large_file.
===================================================================
===================================================================
PROJECT EXECUTION FAILED

galak, 2017-11-29:
I'm guessing this is related to needing to configure CONFIG_FS_NFFS_NUM_CACHE_BLOCKS & CONFIG_FS_NFFS_NUM_CACHE_INODES.  If you look at tests/subsys/fs/nffs_fs_api/testcase.yaml you'll see those set as part of the cache test.

nashif, 2017-11-30:
you need to set additional configs per the testcase.yaml for the cache test

closed 2017-11-30 by nashif