zephyrproject-rtos/zephyr #5207: Bluetooth subsystem uses acl_in_pool even for controllers not supporting flow control

vakulgarg, 2017-11-30:
If zephyr compiled with CONFIG_BT_HCI_ACL_FLOW_CONTROL=y is used with a controller which does not support hci acl flow control, it should not use acl_in_pool. The function bt_buf_get_rx() misses checking the flow control capability of controller and allocates from acl_in_pool for all types which are other than BT_BUF_EVT.

jhedberg, 2017-11-30:
See my comment in the pull request. I'm failing to understand how the current behavior is "bad". Btw, the general assumption is that an application writer fine-tunes the configuration for the hardware that is being targeted, so in this case you simply wouldn't enable CONFIG_BT_HCI_ACL_FLOW_CONTROL since you know your hardware doesn't support it. Don't we also have board-specific defaults which could set this to "n" for your board?

vakulgarg, 2017-12-01:
I now understand that this is a non issue.