zephyrproject-rtos/zephyr #5244: UART continuous interrupt

mhudec1, 2017-12-03:
Hello,

I'm trying to program simple applicatio where when interrupt on uart rx occure, I just read the input buffer and send it back. My problem is, when i send more than 1 Byte, It stops responding and the interrupt is occuring continuously. In case i send only one byte it is working.

Here is the code:
```
static void message_test(char *s,uint8_t *data,uint8_t len)
{	
        int i;
	printk("%s: ",s);
	for(i = 0; i < len; i++)
	{
		printk("%02x ",data[i]);
	}
	printk("(%u bytes)\n",len);
}
static void uart2_isr()
{
	uart_irq_update(uart2_device);
	if(uart_irq_rx_ready(uart2_device))
	{
		uint8_t len = uart_fifo_read(uart2_device,rx_buffer,BUF_MAXSIZE);
		message_test(_func_,rx_buffer,len);
	}	
}
static void uart_init(void)
{
	uart2_device = device_get_binding("UART_2");
	uart_irq_callback_set(uart2_device,uart2_isr);
	uart_irq_rx_enable(uart2_device);
}
```

I tried to code it using example nfc_hello, but there is no uart_irq_update funcion used, if I understood it correctly it is necessary as it is written in documentation.

Maybe this is stupid question, but form is not clear how to do this simple thing.

Best Regards
Michal Hudec

pfalcon, 2017-12-05:
Please see if https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/console/echo/src/main.c would give some ideas/insight.

galak, 2017-12-05:
What hardware/platform are you seeing this on?

mhudec1, 2017-12-05:
stm discovery f3

mhudec1, 2017-12-12:
I tried to done it more ways, but with the same result. The uart_fifo_read() returns whole message I send on uart? For ex. if I send 4 bytes, uart_fifo_read() returns 4 and put those four bytes in buffer? Because this not working for me, I only get 2 bytes shown from 4 and then continouos interrupt....

@pfalcon -> I  tried to implemet it like in getchar.c but with the same result. In the final, I need to implement modbus rtu protocol, but only way what uart is now working is polling....

erwango, 2017-12-12:
Hi,

Can you try with following fix?
https://github.com/zephyrproject-rtos/zephyr/pull/5299

Erwan

nashif, 2018-02-20:
is this already fixed?