zephyrproject-rtos/zephyr #5265: ROM size increase due Zephyr compile options not stripping unused functions

jimmy-huang, 2017-12-05:
When building 3rd party libraries for A101, when CMake file uses  zephyr_get_compile_options_as_string(options) to include compile options, it seemed the it has changed and no longer include some of the flags to strip out unused functions, and therefore, the final ROM image size is increased.

For example,
with latest master, the compile options returned is:
-Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -ffreestanding -include /home/jimmy/projects/zephyrjs/zephyr.js/outdir/arduino_101/zephyr/include/generated/autoconf.h -nostdinc

But with an ealier commit b678895a6 is:
-Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -ffreestanding -include /home/jimmy/projects/zephyrjs/zephyr.js/outdir/arduino_101/zephyr/include/generated/autoconf.h -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -fno-stack-protector -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -nostdinc -Werror=implicit-int -ffunction-sections -fdata-sections -mpreferred-stack-boundary=2 -mno-sse -march=lakemont -mtune=lakemont -msoft-float

SebastianBoe, 2017-12-05:
tl;dr;
This is a regression from f71c3fa6 that is fixed in PR https://github.com/zephyrproject-rtos/zephyr/pull/5229. You can either cherry-pick the PR or wait until it is merged.

The longer story ...

A patch was introduced that increased the use of COMPILE_LANGUAGE generator expressions. COMPILE_LANGUAGE generator expressions were not supported by the zephyr_get_* API until 
PR https://github.com/zephyrproject-rtos/zephyr/pull/5229

In hindsight I should have done both patches at the same time.

SebastianBoe, 2017-12-05:
If you have CONFIG_CPLUSPLUS enabled then this can be closed as a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/5152

grep CONFIG_CPLUSPLUS build_directory/zephyr/.config

SebastianBoe, 2017-12-05:
Oh, and it is necessary to port to a new interface:
https://github.com/zephyrproject-rtos/zephyr/pull/5229/files#diff-c31c4dbe0e95f5d6cc91df73faf59217R17

jimmy-huang, 2017-12-05:
@SebastianBoe Yeah, I do indeed have CONFIG_CPLUSPLUS set,  thanks.

pfalcon, 2017-12-06:
We have MicroPython build fail for arduino_101 https://ci.linaro.org/view/lite-iot-ci/job/lite-aeolus-micropython/189/PLATFORM=arduino_101,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/console due to:

> 00:01:28.150 /home/buildslave/srv/toolchain/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/libexec/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/real-ld: region `ROM' overflowed by 61800 bytes

This didn't happen before. Let me reopen this ticket for further investigation.

pfalcon, 2017-12-06:
CONFIG_CPLUSPLUS is not used.

SebastianBoe, 2017-12-06:
IMHO one shouldn't open issues in Zephyr when external projects have build failures.

A build failure after an update could be due to many reasons ...

Example: https://github.com/zephyrproject-rtos/zephyr/issues/5297

pfalcon, 2017-12-06:
Really? Indeed, it could be due to many reasons, and 90% of those reasons are regressions in the Zephyr build system.

pfalcon, 2017-12-06:
So, Makefile.exports has:

> CC="/home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-gcc"
> Z_CFLAGS=-isystem/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/include -isystem/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/include-fixed-I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/kernel/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/x86/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/x86/soc/intel_quark/quark_se -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/x86/soc/intel_quark/quark_se/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/arch/x86/soc/intel_quark/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/boards/x86/arduino_101 -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/drivers -I/home/pfalcon/projects-3rdparty/micropython/ports/zephyr/outdir/arduino_101/zephyr/include/generated -I/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/include -I/mnt/hdd/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/lib/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/include-fixed -I/home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/iamcu-zephyr-elfiamcu/usr/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/ext/hal/qmsi/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/ext/hal/qmsi/drivers/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/ext/hal/qmsi/soc/quark_se/include-DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DENABLE_EXTERNAL_ISR_HANDLING -DQM_LAKEMONT -DSOC_SERIES=quark_se-Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -ffreestanding -include /home/pfalcon/projects-3rdparty/micropython/ports/zephyr/outdir/arduino_101/zephyr/include/generated/autoconf.h -std=c99 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -nostdinc -Werror=implicit-int -ffunction-sections -fdata-sections -mpreferred-stack-boundary=2 -mno-sse -march=lakemont -mtune=lakemont -msoft-float

I.e., both -ffunction-sections -fdata-sections are there, and actually used when compiling both app and Zephyr code. 

And link is:

> ccache /home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-gcc    CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr_prebuilt.elf -T/home/pfalcon/projects-3rdparty/micropython/ports/zephyr/outdir/arduino_101/zephyr/linker.cmd -Wl,-Map=/home/pfalcon/projects-3rdparty/micropython/ports/zephyr/outdir/arduino_101/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -e__start -Wl,--start-group -Wl,--whole-archive ../libapp.a libzephyr.a lib/libc/newlib/liblib__libc__newlib.a boards/x86/arduino_101/libboards__x86__arduino_101.a ext/hal/qmsi/libext__hal__qmsi.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -Wl,--end-group -L/home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/iamcu-zephyr-elfiamcu/usr/lib/i586-zephyr-elfiamcu/6.2.0 -L/home/pfalcon/projects-3rdparty/micropython/ports/zephyr/outdir/arduino_101/zephyr -lgcc /home/pfalcon/projects-3rdparty/micropython/ports/zephyr/libmicropython.a -nostartfiles -nodefaultlibs -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -lm -lc -L/home/pfalcon/opt/zephyr-sdk-0.9.2/sysroots/iamcu-zephyr-elfiamcu/usr/lib/. -lgcc 

I.e.  -Wl,--gc-sections is there too. Weird.

pfalcon, 2017-12-06:
Prooflink that it worked just 2 days ago: https://ci.linaro.org/view/lite-iot-ci/job/lite-aeolus-micropython/186/PLATFORM=arduino_101,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/console

SebastianBoe, 2017-12-06:
I guess it doesn't link, and therefore you can't run rom_report to find out what is using so much space?

Perhaps you could compare what <toolchain>-size says before and after the update.

pfalcon, 2017-12-06:
@SebastianBoe : I'm on it (and not asking you to look into it), just appreciate being able to investigate within the scope of the upstream project, and within existing ticket, to keep other parties in loop. (Of course, will appreciate your attention when I have come to some results where I can't easily proceed forward - like you always helped so far, thanks!)

pfalcon, 2017-12-06:
@SebastianBoe : It was smart to change this line: https://github.com/zephyrproject-rtos/zephyr/pull/5229/files#diff-c31c4dbe0e95f5d6cc91df73faf59217R20 , and I hope you'd agree it's easy to overlook ;-). Thanks for you patience!

closed 2017-12-06 by pfalcon