zephyrproject-rtos/zephyr #5266: Ensure the Licensing page is up-to-date for the release

Shilpashree3lc, 2017-12-05:
Ensure the "Licensing of Zephyr Project Components" page is updated and correct for the release. See https://www.zephyrproject.org/doc/1.7.0/LICENSING.html.

Shilpashree3lc, 2017-12-05:
Quick review show the below list of files are NOT having any license.

tests/net/socket/tcp/src/main.c
boards/arm/stm32l476g_disco/pinmux.c
samples/basic/threads/src/main.c

pfalcon, 2017-12-05:
> Quick review show the below list of files are NOT having any license.

Which would mean they're licensed under the project license...

Let us know how you'd like to get those resolved - original authors submit patches, or you'd fix that handful of cases on your side. @askawu : FYI, re: tests/net/socket/tcp/src/main.c

nashif, 2017-12-05:
> Which would mean they're licensed under the project license...

true, but we need the header including any copyrights for every file to remove ambiguity

PR #5281 (2017-12-05): Add missing license and copyright headers
The following files didn't have any copyright or license headers on them
when they got contributed.  So add the SPDX Apache license and
appropriate copyright info:

boards/arm/stm32l476g_disco/pinmux.c
samples/basic/threads/src/main.c
tests/net/socket/tcp/src/main.c

Fixes #5266
closed 2017-12-05 by galak