zephyrproject-rtos/zephyr #5274: Issue with pinmux combination for FRDK-K64F setup for cc2520

rveerama1, 2017-12-05:
Pinmux combination in "samples/net/common/cc2520_frdm_k64f.c" failed to work on FRDK-K64F board when both Ethernet and 15.4 are enabled. CC2520 unable to transmit the packets. Specially these two pins.
#define CC2520_GPIO_CCA		17  /* PTC17 */
#define CC2520_GPIO_SFD		16  /* PTC16 */

tbursztyka, 2017-12-05:
As a side note, we tried replacing these 2 pins with PTA1 and PTB9 and it resulted in same issue. It works when ethernet is disabled, and does not when it's enabled. (though according to datasheet - unless I misread it - ethernet has no relationship with these pins at all)

MaureenHelm, 2017-12-05:
PTC16 and PTC17 are used for Ethernet, so you can't use them as GPIO when Ethernet is enabled.

I don't see a conflict for PTA1 or PTB9 unless you've got MCR20A enabled. Can you please provide the code you used to enable them?

rveerama1, 2017-12-07:
[pinmux-patch.log](https://github.com/zephyrproject-rtos/zephyr/files/1538767/pinmux-patch.log)
[config.log](https://github.com/zephyrproject-rtos/zephyr/files/1538770/config.log)

rveerama1, 2017-12-07:
Attached pinmux difference patch and generated config log also (@note: just renamed them to ".log" to satisfy github as an attachment :))

rveerama1, 2017-12-11:
@MaureenHelm can you please look into the attachments?

MaureenHelm, 2017-12-11:
Hi @rveerama1, I looked at the attachments and haven't spotted any problems, but I don't have a CC2520 to reproduce your hardware setup. Given it works with ethernet disabled, I think your board is probably wired correctly and therefore suspect a software problem.

Can you check with a scope that there is a rising edge on SFD or FIFOP when ethernet is enabled? Can also you step through the code to verify that the gpio callbacks get installed in this configuration, and see if you ever execute the callbacks? Have you tried enabling SYS_LOG_DBG in the CC2520 driver?

tbursztyka, 2018-02-13:
Hi @MaureenHelm, I just tried mcr20a since the shield is meant to work of frdm directly and the issue happens also unfortunately :(   looks like it's unable to do CCA (the "irqb" never fires, aka: is never caughn - so it times-out)

If I disable the ethernet port on the board: it works properly.

I am a bit lost, none of the pins are shared with etherent, afaik. Pinmuxing looks ok also.

MaureenHelm, 2018-02-26:
> @MaureenHelm @dleach02 Any updates on this?

Hi @rveerama1. I spent some time last week debugging this, but didn't find anything yet. Will continue today.

rveerama1, 2018-02-27:
@MaureenHelm Thanks, let me know if you have any fix.

MaureenHelm, 2018-03-02:
> I just tried mcr20a since the shield is meant to work of frdm directly and the issue happens also unfortunately :( looks like it's unable to do CCA (the "irqb" never fires, aka: is never caught - so it times-out)

I checked with a scope, and the irqb signal asserts just once, during mcr20a_init before the gpio callback is configured, and never asserts again. I can see all of the spi transactions in power_on_and_setup() complete on a scope. This behavior is the same with and without ethernet enabled.

I've also tried running tests/drivers/spi/spi_basic_api with MOSI/MISO shunted, with and without ethernet enabled, and it passes in both cases.

I'm not so sure this is a pinmuxing issue.

tbursztyka, 2018-03-06:
@MaureenHelm The issue is not spi related, at all. It's about the GPIO interrupt. But on mrc20a there seeems to be something else to it, timing related that makes cca failing if debugging output level is low.

rveerama1, 2018-04-04:
@MaureenHelm @jfischer-phytec-iot @tbursztyka To reproduce this issue on frdm_k64f + mcr20a (15.4 radio) https://github.com/zephyrproject-rtos/zephyr/pull/5035/.
sample application is : samples/net/rpl_border_router, conf_file = prj_frdm_k64f_mcr20a.conf.

tbursztyka, 2018-04-05:
Ok latest news: mcr20a just works fine with frdm k64f. It happens that @rveerama1 has a bogus hardware finally (either the mcr20a or the frdm board, we don't know). But testing the rpl border router sample works totally fine on my hardware. 

I might retry cc2520 at some point, just to make sure. After all, mcr20a did not work before for some reason (cca sequencing was always failing _but_ in full debug mode...).

nashif, 2018-05-15:
@rveerama1 ping

rveerama1, 2018-05-16:
@MaureenHelm yes, I received new hardware (mcr20a) and it works (except GPIO issue which I raised). But original issue still exists "pinmux combination for FRDK-K64F setup for cc2520". Right pins for cc2520 add on board in case of Ethernet also enabled ?
or shall we drop the combination of frdm_k64f + cc2520 add on board support ?

tbursztyka, 2018-05-16:
why dropping? we could just raise an #error message in  relevant file if ethernet is enabled too.

rveerama1, 2018-05-17:
RPL BR sample experimented with frdm_k64f + cc2520 and frdm_k64f + mcr20a board combinations. Encountered issue with frdm_k64f + cc2520 as cc2520 share common gpio pins with Ethernet for Tx and Rx. Tried different combinations of other available pins, but cc2520_cca() was always busy. Could not proceed further. Dropping this combination.
frdm_k64f + mcr20a combination has better results.

closed 2018-05-17 by rveerama1