zephyrproject-rtos/zephyr #5283: reference to non-existing functions in arch/x86/core/intstub.S

nashif, 2017-12-05:
```

 * Only the volatile integer registers are saved since ISRs are assumed not to
 * utilize floating point (or SSE) instructions.  If an ISR requires the usage
 * of floating point (or SSE) instructions, it must first invoke nanoCpuFpSave()
 * (or nanoCpuSseSave()) at the beginning of the ISR.  A subsequent
 * nanoCpuFpRestore() (or nanoCpuSseRestore()) is needed just prior to returning
 * from the ISR.  Note that the nanoCpuFpSave(), nanoCpuSseSave(),
 * nanoCpuFpRestore(), and nanoCpuSseRestore() APIs have not been
 * implemented yet.

```

most of the above functions were probably renamed and do not exist, need to clean this up and use current implementation functions in the docs.

dbkinder, 2017-12-11:
Note the mentioned documentation says these APIs have not been implemented yet, so perhaps they never have, and we should just delete this paragraph except for the first sentence?

carlescufi, 2018-02-06:
@andrewboie could you give us the names of the functions that replace the former `nano*` ones?

PR #6494 (2018-03-06): arch: x86: fix integer stub comments
The comment was obsolete; we simply do not allow use of the FPU or
vector math in ISRs. There is no desire to add such support, doing
this is properly offloaded to a worker thread.

Fixes #5283.
commit c0771ea2 (2018-03-06): arch: x86: fix integer stub comments

The comment was obsolete; we simply do not allow use of the FPU or
vector math in ISRs. There is no desire to add such support, doing
this is properly offloaded to a worker thread.

Fixes #5283.
closed 2018-03-06 by nashif