zephyrproject-rtos/zephyr #5284: update altera_max10 board documentation and remove references to nanokernel

nashif, 2017-12-05:
in the board documentation for max10 we still talk about nanokernel and show an example with gdb using the old kernel files, this needs to be updated:

```
  (gdb) b nano_init
   Breakpoint 1 at 0x1de8: file /projects/zephyr2/kernel/nanokernel/nano_init.c, line 207.
   (gdb) c
   Continuing.

   Breakpoint 1, _Cstart () at /projects/zephyr2/kernel/nanokernel/nano_init.c:323
   323      nano_init((struct tcs *)&dummyTCS);
   (gdb) bt
   Python Exception <type 'exceptions.ImportError'> No module named gdb.frames:
   #0  _Cstart () at /projects/zephyr2/kernel/nanokernel/nano_init.c:323
   #1  0x00003ca4 in _PrepC () at /projects/zephyr2/arch/nios2/core/prep_c.c:61
   #2  0x00003ccc in __start () at /projects/zephyr2/arch/nios2/core/crt0.S:155

```

ramakrishnapallala, 2017-12-06:
PR submitted:
https://github.com/zephyrproject-rtos/zephyr/pull/5301

nashif, 2017-12-06:
> PR submitted:
> #5301

thanks, you should always do the reference in the PR itself, for example:

"Fixes #5284", github then will pick this up and automate closing the issue

PR #5301 (2017-12-06): boards: nios2: altera_max10: cleanup board documentation
Remove references to nanokernel from the board documentation.

Fixes #5284
closed 2017-12-06 by nashif