zephyrproject-rtos/zephyr #5286: NET_L2: Symbols from the L2 implementation layer are exposed to users of L2

SebastianBoe, 2017-12-05:
Hi,

for many reasons, which I assume I don't need to go into here, users of an interface should not
have access to details of it's implementation.

The L2 layer violates this principle in net_l2.h when it includes ieee802154.h:

```
#ifdef CONFIG_NET_L2_DUMMY
#define DUMMY_L2		DUMMY
#define DUMMY_L2_CTX_TYPE	void*
NET_L2_DECLARE_PUBLIC(DUMMY_L2);
#endif /* CONFIG_NET_L2_DUMMY */

#ifdef CONFIG_NET_L2_ETHERNET
#define ETHERNET_L2		ETHERNET
#define ETHERNET_L2_CTX_TYPE	void*
NET_L2_DECLARE_PUBLIC(ETHERNET_L2);
#endif /* CONFIG_NET_L2_ETHERNET */

#ifdef CONFIG_NET_L2_IEEE802154
#include <net/ieee802154.h>
#define IEEE802154_L2		IEEE802154
#define IEEE802154_L2_CTX_TYPE	struct ieee802154_context
NET_L2_DECLARE_PUBLIC(IEEE802154_L2);
#endif /* CONFIG_NET_L2_IEEE802154 */

#ifdef CONFIG_NET_L2_BT
#define BLUETOOTH_L2		BLUETOOTH
#define BLUETOOTH_L2_CTX_TYPE	void*
NET_L2_DECLARE_PUBLIC(BLUETOOTH_L2);
#endif /* CONFIG_NET_L2_BT */

#ifdef CONFIG_NET_OFFLOAD
#define OFFLOAD_IP_L2		OFFLOAD_IP
#define OFFLOAD_IP_L2_CTX_TYPE	void*
NET_L2_DECLARE_PUBLIC(OFFLOAD_IP);
#endif /* CONFIG_NET_OFFLOAD */

```

Presumably, it is including ieee802154.h because it wants to expose struct ieee802154_context.

I don't know why it is doing this, but I hope it is possible to either;

Use a void* pointer like the other implementations do. Presumably they have already solved the same problem as ieee80215?

Split up ieee802154.h such that as little as possible is exposed to the application code.

tbursztyka, 2017-12-05:
NET_DEVICE_INIT needs to know the type in order to allocate the space relevantly, in the right dedicated section in ROM &/or RAM. There is nothing violated here. (Other solution could be made of just a "size" and it would allocate a u8_t buffer of that size, which in turn would be casted in the right place to the right context type. But it is as hackish as what we have now, and actually a bit crappier imo)

Other L2 don't have a context, but BT which in case is quite particular as the L2 itself instantiates the unique net_if (and instead, it uses device's private data to do so). That's why only 15.4 has one here.

net_l2.h is all but application oriented btw, it's targeting net stack L2 and drivers. Plus, "L2 users", the drivers, could access that context. Nothing prevents that from the current design, the need hasn't shown up yet though.

That said, the net_mgmt part in ieee80215.h could be put into ieee802154_mgmt.h, why not. That part at least is used by application, and these ones have nothing to do with the context. But that's not the point of this current issue, which should be closed.

closed 2017-12-05 by tbursztyka

SebastianBoe, 2017-12-06:
> net_l2.h is all but application oriented btw, it's targeting net stack L2 and drivers. Plus, "L2 users", the drivers, could access that context. Nothing prevents that from the current design, the need hasn't shown up yet though.

That is what I thought as well, so I was confused when I saw symbols from ieee802154.h being accessable from a sample. Specifically, echo_server's udp.c will transitively include ieee802154.h.

I'm not familiar with the networking stack, but I did not expect a UDP user like echo_server/src/udp.c to have access to ieee80215.h symbols.

Thank you for the detailed response, I will try to work around this now that you have confirmed it is not just some temporary hack.

tbursztyka, 2017-12-06:
I am actually moving the net mgmt out of that file into ieee802154_mgmt.h, it's indeed better to split up these at least.