zephyrproject-rtos/zephyr #5297: Build failure issue on latest Zephyr commit

kumarvikash1, 2017-12-06:
Build broken by additional build flags, details given below

commit abbdd88683abddb44197abcd5acd8a59e8fa10b5
Author: Sebastian Bøe <sebastian.boe@nordicsemi.no>
Date:   Tue Dec 5 12:24:01 2017 +0100

    cmake: Fix EXTRA_LDFLAGS feature

    It is supported to add give extra flags to the linker from the
    commandline like this:

    cmake -DEXTRA_LDFLAGS=-Lmy_dir path

    But unfortunately this was broken during the CMake
    migration. Interestingly, the reason that it was broken is that KBuild
    was also partially broken. KBuild would pass on EXTRA_LDFLAGS when
    object files were linked together into built-in.o files, but it would
    not use EXTRA_LDFLAGS for the final link into an elf file.

    This patch fixes EXTRA_LDFLAGS.

    Signed-off-by: Sebastian Bøe <sebastian.boe@nordicsemi.no>

SebastianBoe, 2017-12-06:
Could not reproduce.

Please post steps to reproduce.

Board, sample, CONF_FILE etc.

SebastianBoe, 2017-12-06:
Note that this can expose pre-existing problems in application code.

What this patch does is ensure that EXTRA_LDFLAGS are given to the linker.

So if a user had, say, --fatal-warnings in his EXTRA_LDFLAGS, then before this patch
he could have a warning in his application without it causing a build failure.

kumarvikash1, 2017-12-06:
Issues is : 
build output: i586-zephyr-elf-gcc: error: unrecognized command line option ‘--fatal-warnings’

revert works fine;
git revert abbdd88683abddb44197abcd5acd8a59e8fa10b5

It has failed on almost all boards, 
like : qemu_x86, qc_1000 etc....

Hope it helps.

SebastianBoe, 2017-12-06:
Could not reproduce.

I need the command you are running, the environment variables you are using, and the git commit revision of a clean git repo.

I can't see any reference to --fatal-warnings, only -Wl,--fatal-warnings, so I suspect there is something
on your end that is needed to reproduce the issue.

What you could do is grep through your environment variables and your repo for --fatal-warnings to
find out where it is coming from.

sebo@mach:~/zephyr$ ack "\--fatal-warnings"
scripts/sanitycheck
680:\t\t-DEXTRA_AFLAGS=-Wa,--fatal-warnings\\
681:\t\t-DEXTRA_LDFLAGS=-Wl,--fatal-warnings\\

kumarvikash1, 2017-12-06:
zephyr.git/tests/crypto/ctr_prng/testcase.yaml#test

Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit ID                                Branch          URL
---------------------------------------- --------------- -------------------
7d0ab1b96abaab502c3722f44aa852d2af4e7526 master          https://github.com/zephyrproject-rtos/zephyr.git

Detailed description
====================

configure passed
build passed: 'mkdir -p h5bc/outdir-h5bc-konu-qemu_riscv32; rm -f /outdir-h5bc-konu-qemu_riscv32/.config'
build passed: 'cmake -DPREBUILT_HOST_TOOLS=/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin -DBOARD=qemu_riscv32 -DARCH=riscv32 -DEXTRA_CPPFLAGS="-DTC_RUNID=ci-171205-1931-658:konu" -DEXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" -DEXTRA_AFLAGS=-Wa,--fatal-warnings -DEXTRA_LDFLAGS=--fatal-warnings  -B"/outdir-h5bc-konu-qemu_riscv32" -H"//tests/crypto/ctr_prng"'
build output: CMake Deprecation Warning at /LABEL/verify/SHARD/2-3/ZEPHYR_GCC_VARIANT/zephyr/zephyr.git/cmake/app/boilerplate.cmake:47 (cmake_policy):
build output: Call Stack (most recent call first):
build output:   CMakeLists.txt:1 (include)
build output: Built target config-sanitycheck
build output: [ 95%] Linking C executable zephyr_prebuilt.elf
build output: riscv32-zephyr-elf-gcc: error: unrecognized command line option â€˜--fatal-warningsâ€™; did you mean â€˜--all-warningsâ€™?
build output: zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:100: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
build output: make[3]: *** [zephyr/zephyr_prebuilt.elf] Error 1
build output: CMakeFiles/Makefile2:568: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
build output: make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
build output: Makefile:83: recipe for target 'all' failed
build output: make[1]: *** [all] Error 2
build failed

SebastianBoe, 2017-12-06:
Your test infrastructure is calling cmake like this:

cmake -DPREBUILT_HOST_TOOLS=/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin -DBOARD=qemu_riscv32 -DARCH=riscv32 -DEXTRA_CPPFLAGS="-DTC_RUNID=ci-171205-1931-658:konu" -DEXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" -DEXTRA_AFLAGS=-Wa,--fatal-warnings -DEXTRA_LDFLAGS=--fatal-warnings -B"/outdir-h5bc-konu-qemu_riscv32" -H"//tests/crypto/ctr_prng"'

You need to change your infrastructure from using "--fatal-warnings" to using "-Wl,--fatal-warnings"

I can give more details if this unclear.

closed 2017-12-06 by SebastianBoe

rishikha, 2017-12-06:
Updated the test infra by the same