zephyrproject-rtos/zephyr #5298: Endless loop in uart pipe

erwango, 2017-12-06:
Reported in devel mailing list by Dmitri Shmidt.

"The problem is in uart_pipe_isr() logic. It is never-ending (spinning)
after first time entrance: it is still processing RX but its nature doesn't allow any user
threads to get control."

https://lists.zephyrproject.org/pipermail/zephyr-devel/2017-December/008533.html

pfalcon, 2017-12-06:
Note that I saw similar effects in https://github.com/zephyrproject-rtos/zephyr/pull/42, and while working on irq-driven console in general: https://github.com/zephyrproject-rtos/zephyr/pull/49.

erwango, 2017-12-06:
> while working on irq-driven console in general: #49.

Wouldn't it be because there is always a tx associated to rx in console case?
Hence TXE is set to 0 and while loop is exited.
I have been to reproduce issue by commenting out insert_char() in  uart_console_isr, but no issue as is.

pfalcon, 2017-12-06:
Sure, for the console case it was solved by following invariants later documented in https://github.com/zephyrproject-rtos/zephyr/pull/159 , https://github.com/zephyrproject-rtos/zephyr/pull/281. And I had to patch existing test in #42 to follow them too.

So, my points is that such problems were known, and there's an existing pattern how to resolve them. And my concern that this case wasn't resolved in a way to break previous cases. (I'm going to look at #5299 in detail, from the title, it sounds good).

PR #5299 (2017-12-06): drivers: serial: stm32: report only unmasked irq
Fix #5298
irq_is_pending function returned TXE/RXNE flag status
even if IRQ was masked, which led to enless loop
in uart_pipe when no TX was performed. Fix by reporting status only when
IRQ is unmasked.
commit 222bc609 (2017-12-06): drivers: serial: stm32: report only unmasked irq

Fix #5298
irq_is_pending function returned TXE/RXNE flag status
even if IRQ was masked, which led to enless loop
in uart_pipe when no TX was performed. Fix by reporting status only when
IRQ is unmasked.
closed 2017-12-06 by galak