zephyrproject-rtos/zephyr #5308: uart_pipe rx-spinning problem for stm32

krisik70, 2017-12-06:
There is a problem in uart_pipe_isr(). It is never-ending (spinning) after first time entrance: it is still processing RX but its nature doesn't allow any user threads to get control.
uart_pipe_isr() is called on RX interrupt:
static void uart_pipe_isr(struct device *unused) {
         while (uart_irq_update(uart_pipe_dev)
                   && uart_irq_is_pending(uart_pipe_dev)) {
                ...
         }

uart_irq_update() is always 1 and uart_irq_is_pending()
is also always 1 (in case of stm32):
         return (LL_USART_IsActiveFlag_RXNE(UartInstance) ||
                     LL_USART_IsActiveFlag_TXE(UartInstance));

Erwan proposed to mask TXE if it is not enabled ,and it is working fine:
static int uart_stm32_irq_is_pending(struct device *dev)
{
    USART_TypeDef *UartInstance = UART_STRUCT(dev);

    return ((LL_USART_IsActiveFlag_RXNE(UartInstance) &&
                LL_USART_IsEnabledIT_RXNE(UartInstance)) ||
               (LL_USART_IsActiveFlag_TXE(UartInstance) &&
                LL_USART_IsEnabledIT_TXE(UartInstance)));
}

galak, 2017-12-06:
Little confused, is this fixed by #5299?

krisik70, 2017-12-06:
Sorry for the noise - it was my miscommunication with Erwan.

closed 2017-12-06 by krisik70

erwango, 2017-12-07:
@krisik70 : My bad actually, I finally did it on my own to get a chance to get it 1.10.
I should have informed you