zephyrproject-rtos/zephyr #5317: IPSP deadlock during disconnect and net_if_down

rsalveti, 2017-12-08:
IPSP disconnect process can cause a deadlock if called without enough free tx_bufs.

Example (application waiting for available tx buffer):
```
[0278490] [net/buf] [DBG] net_buf_alloc_debug: net_pkt_get_reserve_data():356: Pool tx_bufs blocked for 2 secs
[0278500] [net/buf] [DBG] net_buf_alloc_debug: net_pkt_get_reserve_data():356: Pool tx_bufs low on buffers.
```

BT LE disconnect event received:
```
[DEBUG] terminate_ind_rx_enqueue: enter
[DEBUG] disconn_complete: enter
[DEBUG] hci_disconn_complete:
[0278570] [bt] [DBG] bt_conn_set_state: connected -> disconnected
[0278570] [bt] [DBG] bt_l2cap_chan_del: conn 0x2000168c chan 0x200017e0
[0278580] [bt] [DBG] bt_l2cap_chan_del: calling ops->disconnected(chan)
[0278590] [bt] [DBG] bt_att_disconnected: chan 0x200017e0 cid 0x0004
[0278600] [bt] [DBG] bt_l2cap_chan_del: setting chan->conn to NULL
[0278600] [bt] [DBG] bt_l2cap_chan_del: conn 0x2000168c chan 0x2000194c
[0278610] [bt] [DBG] bt_l2cap_chan_del: calling ops->disconnected(chan)
[0278620] [bt] [DBG] bt_smp_disconnected: chan 0x2000194c cid 0x0006
[0278620] [bt] [DBG] bt_l2cap_chan_del: setting chan->conn to NULL
[0278630] [bt] [DBG] bt_l2cap_chan_del: conn 0x2000168c chan 0x20001730
[0278640] [bt] [DBG] bt_l2cap_chan_del: calling ops->disconnected(chan)
[0278650] [bt] [DBG] l2cap_disconnected: ch 0x20001730 cid 0x0005
[0278650] [bt] [DBG] bt_l2cap_chan_del: setting chan->conn to NULL
[0278660] [bt] [DBG] bt_l2cap_chan_del: conn 0x2000168c chan 0x2000ee30
[0278670] [bt] [DBG] bt_l2cap_chan_del: calling ops->disconnected(chan)
[DEBUG] ipsp_disconnected: enter
[0278680] [net/bt] [DBG] ipsp_disconnected: Channel 0x2000ee30 disconnected
[DEBUG] net_if_down: enter
[DEBUG] leave_mcast_all: enter
[DEBUG] net_ipv6_mld_leave: enter
[DEBUG] send_mldv2: enter
```

send_mldv2 causes a deadlock when it tries to append data to the net_pkt:
```
static int send_mldv2(struct net_if *iface, const struct in6_addr *addr,
                      u8_t mode)
{
        struct net_pkt *pkt;
        int ret;
        pkt = net_pkt_get_reserve_tx(net_if_get_ll_reserve(iface, NULL),
                                     K_FOREVER);
        net_pkt_append_be16(pkt, 1); /* number of records */
        ...
```

Deadlock happens because it tries to allocate more buffers from tx_bufs (net_pkt_get_reserve_tx uses tx_pkts by default), when we already have another thread also waiting for available tx_bufs.

Not sure if leave_mcast_all should be called when over IPSP, as at that point the bluetooth connection is already gone, so no point trying to send more net data.

rsalveti, 2017-12-08:
Unable to reproduce the issue after commenting out leave_mcast_all.

rsalveti, 2018-01-23:
@jukkar mind having a look at this issue?

jukkar, 2018-01-23:
You could just disable MLD support in this case as that is probably not needed for BT, just do CONFIG_NET_IPV6_MLD=n

rsalveti, 2018-01-23:
https://tools.ietf.org/html/draft-ietf-6lo-btle-17 says "A 6LN MUST join the all-nodes multicast address. There is no need for 6LN to join the solicited-node multicast address, since 6LBR will    know device addresses and hence link-local addresses of all connected 6LNs." so it I would guess CONFIG_NET_IPV6_MLD is still required, even when using with BT.

The problem is that leave_mcast_all would still get called after the BT connection is already gone. Should we create another flag that is BT specific to avoid calling join_mcast_solicit_node and leave_mcast_all?

jukkar, 2018-01-24:
> "A 6LN MUST join the all-nodes multicast address

Ah, yes indeed we need the MLD support enabled.

> The problem is that leave_mcast_all would still get called after the BT connection is already gone. Should we create another flag that is BT specific to avoid calling join_mcast_solicit_node and leave_mcast_all?

Sounds like a good plan, would you be able to prepare a patch for that?

rsalveti, 2018-01-25:
> Sounds like a good plan, would you be able to prepare a patch for that?

Sure, will try to get a patch ready by tomorrow, thanks!

rsalveti, 2018-01-26:
Created https://github.com/zephyrproject-rtos/zephyr/pull/5846 to skip solicited node when L2 BT.

Now for leave_mcast_all I was trying to find what would be the best way to skip this function when iface->l2 is L2 BT, and follows some options:

```
diff --git a/subsys/net/ip/net_if.c b/subsys/net/ip/net_if.c
index 94a95e8..d25fe98 100644
--- a/subsys/net/ip/net_if.c
+++ b/subsys/net/ip/net_if.c
@@ -1841,7 +1841,16 @@ int net_if_down(struct net_if *iface)

        NET_DBG("iface %p", iface);

+#if defined(CONFIG_NET_L2_BT)
+       /* Skip leave multicast if L2 BLUETOOTH as the L2CAP channel
+        * is in the process of being disconnected.
+        */
+       if (iface->l2 != &NET_L2_GET_NAME(BLUETOOTH)) {
+               leave_mcast_all(iface);
+       }
+#else
        leave_mcast_all(iface);
+#endif

        net_if_flush_tx(iface);

```
This change simply skip leave_mcast_all if iface->l2 == L2 BT, but that doesn't take into account the L2CAP channel connection state. Do we also need to support net_if_down when the L2CAP channel is still active?

The second option would be checking the connection state before calling leave_mcast_all, but that would require changes in L2CAP (as the state is only set after the disconnect callback gets called) and additional logic to fetch the channel and verify the state, which might end up adding too much L2 BT specifics in net_if.c, so not sure if desired.

Another possible option would be adding an extra argument to net_if_down that could indicate if the L2 stack is still valid/functional, and use that to decide when to call leave_mcast_all. That would avoid adding L2 BT specific logic in net_if.c but would mean changing the net_if_down function interface.

What do you think?

jukkar, 2018-01-31:
> This change simply skip leave_mcast_all if iface->l2 == L2 BT, but that doesn't take into account the L2CAP channel connection state. Do we also need to support net_if_down when the L2CAP channel is [...]

Lets try to avoid adding too much BT specific logic to net_if.c so I would not like to add any new parameter to net_if_down().
Your proposal to avoid call to leave_mcast_all() looks fine to me, currently we only call net_if_down() from BT but at that point the channel is already disconnected. It is unlikely that net_if_down() would be called from any other place for BT.

Vudentz, 2018-02-08:
Either we do this or we add a parameter to net_if_down to inform the reason which is then used to infer if leave_mcast_all is needed or not.

Btw, we used to have net_if_down as a means to do carrier detection, so having leave_mcast_all there seems wrong when we know there is no connection to deliver the packets.

commit 656bab03 (2018-02-19): net: Introduce net_if_carrier_down

This introduces net_if_carrier_down so the L2 driver can inform when it
has lost connectivity so all packets shall be flushed and the interface
should be put down.

Fixes #5317
commit ec6188bc (2018-02-19): net: bt: Call net_if_carrier_down when disconnected

If IPSP channel is disconnected call net_if_carrier_down instead of
net_if_down since the later may still attempt to send packets while the
former just discard them immediatelly.

Fixes #5317
closed 2018-02-19 by jukkar