zephyrproject-rtos/zephyr #5327: 1.11 Release Checklist

galak, 2017-12-08:
Add items as required
--------
- [ ] Release Notes WIP
  - [ ] Create draft doc/release-notes-1.11.rst with feature summary from https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update doc/release-notes-1.11.rst details sections (relative to 1.10.0 release) as important issues are addressed (subsystem owners)
--------
- [ ] Release data
  - [ ] Update version in VERSION
  - [ ] Run sanitycheck with --all --release and checkin resulting file scripts/sanity_chk/sanity_last_release.csv
--------
- [ ] Finalize Release Notes
  - [x] Add list of GitHub issues
  - [ ] Review doc/release-notes-1.11.rst overview summary and details sections (see Sign-off below)
  - [x] Update doc/release-notes.rst (add reference to 1.11 notes)
  - [x] Update doc/index.rst (add reference to doc/1.11.0)
  - [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases
  - [ ] Final tagged release as zephyr-v1.11.0 (with release notes)
  - [ ] Update https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update https://www.zephyrproject.org/developers/#downloads
  - [ ] Create branch for 1.11.x releases
  - [ ] Update doc generation (in future should be automatic based on tag)
--------
- [x]  Final Release Notes Sign-off
  - Subsystems
    - [x] Networking - @jukkar 
    - [x] Bluetooth - @jhedberg , @carlescufi 
  - Architectures
    - [x] ARC - @ruuddw 
    - [x] ARM - @galak, @MaureenHelm 
    - [x] X86 - @nashif, @andrewboie 
    - [ ] Xtensa, RISCV32, NIOS-II - @nashif, @andrewboie 
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)

pfalcon, 2018-02-05:
Release notes should include something like: "Cleanups and consistency improvements in sensor subsystem: SENSOR_CHAN_DISTANCE is changed to use meter unit, SENSOR_CHAN_HUMIDITY in percent units (backward incompatible changes)". I can prepare specific wording when release notes will be drafted.

carlescufi, 2018-03-09:
Next release: https://github.com/zephyrproject-rtos/zephyr/issues/6533

MaureenHelm, 2018-03-13:
@carlescufi Why did you add the v1.12.0 milestone? Can we close this issue?

carlescufi, 2018-03-13:
@MaureenHelm my mistake. Closing now.

closed 2018-03-13 by carlescufi