zephyrproject-rtos/zephyr #5328: build system should try and create conf if not found for non-zephyr SDK

galak, 2017-12-08:
if we are using something like gcc arm embedded toolchain, we shouldn't need to have the zephyr sdk installed and set ZEPHYR_SDK_INSTALL_DIR to find things like 'conf'.  So for 'conf' we should attempt to build it from the source tree if its not found.

nashif, 2017-12-08:
We should just document this if not already, IMO this should not be part of the build process. On the mac for example what I did was build it once, put it in my path and forget about it...

SebastianBoe, 2017-12-11:
> if we are using something like gcc arm embedded toolchain, we shouldn't need to have the zephyr sdk installed and set ZEPHYR_SDK_INSTALL_DIR to find things like 'conf'.

It is not necessary to have the SDK installed. The docs describe that what nashif is describing
is supported.

closed 2017-12-11 by SebastianBoe