zephyrproject-rtos/zephyr #5343: cmake: libapp.a in unexpected location

andrewboie, 2017-12-11:
I notice that libapp.a is directly in the build directory instead of being under zephyr/ with the rest of the build artifacts.
Is this intentional?

```
apboie@wintermute:/projects/zephyr2/samples/philosophers/out master /projects/zephyr2
$ ls -l
total 256
-rw-rw-r--.  1 apboie apboie  19938 Dec 11 13:29 CMakeCache.txt
drwxrwxr-x. 12 apboie apboie   4096 Dec 11 13:29 CMakeFiles
-rw-rw-r--.  1 apboie apboie   1566 Dec 11 13:29 cmake_install.cmake
-rw-rw-r--.  1 apboie apboie 184762 Dec 11 13:29 libapp.a
-rw-rw-r--.  1 apboie apboie  37935 Dec 11 13:29 Makefile
drwxrwxr-x. 15 apboie apboie   4096 Dec 11 13:29 zephyr
```

SebastianBoe, 2017-12-12:
This was unintentional, it would have been cleaner to have it somewhere in <build>/zephyr/**

SebastianBoe, 2018-01-04:
Oops, I referenced this issue by mistake in #1c2de10.

SebastianBoe, 2018-08-06:
This patch resolves the issue: https://github.com/SebastianBoe/zephyr/commit/d3fa360f4d6970c129c7476bdc1060e8eb60788f

But I am hesitant to apply it as it is not obvious to me whether the improved organization outweighs the possibility of breaking backwards compatibility with external systems.

galak, 2018-12-06:
@SebastianBoe I think we should go ahead with this change.  Can you please submit a PR for this.

SebastianBoe, 2018-12-07:
>  Can you please submit a PR for this.
-- @galak 

Done.

commit 1c2de10c (2018-01-03): cmake: ninja: Change how CMake names the kernelspace archives

The linker script places kernelspace and userspace archives in
different sections. But the linker script itself does not determine
what archives are in what space, that is done by CMake.

CMake passes the list of kernelspace archives to the linker script
through defines, like this:

-DNUM_KERNEL_OBJECT_FILES=3
-DKERNEL_OBJECT_FILE_0=path/to/archive_a.a
-DKERNEL_OBJECT_FILE_1=path/to/archive_b.a
-DKERNEL_OBJECT_FILE_2=path/to/archive_c.a

These paths are relative, and since Ninja and Make invoke the linker
with different "working di [... cut]
PR #11930 (2018-12-07): cmake: Move libapp.a into it's own directory
Move libapp.a into it's own directory. This would make it's placement
consistent with how zephyr.elf is placed.

E.g. it might now look like:

ls b/*

b/build.ninja  b/CMakeCache.txt  b/cmake_install.cmake  b/rules.ninja

b/app:
libapp.a

b/CMakeFiles:
3.12.0   cmake.check_cache  CMakeOutput.log  TargetDirectories.txt
app.dir  CMakeError.log     CMakeTmp

b/zephyr:
arch                 kernel                       subsys
boards               lib                          tests
cmake                liboffsets.a                 zephyr.bin
CMakeFiles    [... cut]
closed 2018-12-07 by nashif