zephyrproject-rtos/zephyr #5344: samples/net/http_client: unable to send the proper http request to Apache server  in IPv6

gurucharanx, 2017-12-12:
The http_client sample application is unable to send a proper http request to the Apache server due to which the server replies with bad request
The test was done for IPv6
The steps are as followed in #4324 except the configuration setting change in prj.conf file as below

CONFIG_NET_IPV4=n
CONFIG_NET_IPV6=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_DEBUG_HTTP=n
CONFIG_NET_APP_NEED_IPV6=y
CONFIG_NET_APP_NEED_IPV4=n

The console output (at qemu) is as below:

```
_Scanning dependencies of target run
[100%] 
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32,+nx,+pae
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

shell> [http-client] [INF] main: --------Sending 20 sync request--------
[http-client] [INF] do_sync_http_req: [1] Send /index.html
[http-client] [INF] do_sync_http_req: HTTP server response status: Bad Request
[http-client] [INF] do_sync_http_req: HTTP body: 347 bytes, expected: 347 bytes
[http-client] [INF] do_sync_http_req: [1] Send /
[http-client] [ERR] do_sync_http_req: Cannot send HEAD request (-60)
[http-client] [INF] main: Done!
QEMU: Terminated_
```

The test was verified on commit c84a283d6866087ebb95a76515604b7b8bd4455e

gurucharanx, 2017-12-12:
[](url)https://github.com/zephyrproject-rtos/zephyr/issues/4325

pfl, 2017-12-12:
Do we already have a tcpdump output of the issue?

gurucharanx, 2017-12-12:
[http_client_ipv6.txt](https://github.com/zephyrproject-rtos/zephyr/files/1551268/http_client_ipv6.txt)
Please find the attached log of wireshark for this issue. Please rename the file before using it.

rveerama1, 2017-12-19:
I can reproduce the issue with http_client. And also with 'telnet' too.
---------------------------------------------------------------------------------------
telnet 2001:db8::2 8000
Trying 2001:db8::2...
Connected to 2001:db8::2.
Escape character is '^]'.
GET / HTTP/1.1
Host: 2001:db8::2

HTTP/1.1 400 Bad Request
Date: Tue, 19 Dec 2017 09:27:12 GMT
Server: Apache/2.4.25 (Ubuntu)
Content-Length: 421
Connection: close
Content-Type: text/html; charset=iso-8859-1

<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>400 Bad Request</title>
</head><body>
<h1>Bad Request</h1>
<p>Your browser sent a request that this server could not understand.<br />
</p>
<p>Additionally, a 400 Bad Request
error was encountered while trying to use an ErrorDocument to handle the request.</p>
<hr>
<address>Apache/2.4.25 (Ubuntu) Server at 127.0.1.1 Port 2</address>
</body></html>
Connection closed by foreign host.
----------------------------------------------------------------------------------------------------

It looks like Apache2 server with IPv6 address [tcp connected, but http request failed] not working or I might be missing some details. Can you try Apache2 server with IPv6 address without zephyr and share some success logs ?

rveerama1, 2018-02-14:
> Apache/2.4.25 (Ubuntu) Server at 127.0.1.1 Port 2 Connection closed by foreign host. ----------------------------------------------------------------------------------------------------
> 
> It looks like Apache2 server with IPv6 address [tcp connected, but http request failed] not working or I might be missing some details. Can you try Apache2 server with IPv6 address without zephyr and share some success logs ?

@gurucharanx any comments?

nashif, 2018-02-15:
@kumarvikash1 can you still reproduce this?

rveerama1, 2018-03-20:
Closing, re-open this issue it it exists.

closed 2018-03-20 by rveerama1