zephyrproject-rtos/zephyr #5345: Cmake: ext: Header file include error: No such file or directory

miyatsu, 2017-12-12:
In hello world sample, do following steps, may get compile error.

Step1: vim prj.conf, add:
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y

Step2: vim src/main.c, add:
#include <mbedtls/sha256.h> above the main function.

Step3:
mkdir -p build/nrf51_pca10028 && cd build/nrf51_pca10028
cmake -DBOARD=nrf51_pca10028 ./ ../../
make -j4

Step4: Output with error:

/************************************************************************/
/home/username/project/zephyr/samples/hello_world/src/main.c:10:28: fatal error: mbedtls/sha256.h: No such file or directory
 #include <mbedtls/sha256.h>
                            ^
compilation terminated.
CMakeFiles/app.dir/build.make:302: recipe for target 'CMakeFiles/app.dir/src/main.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/src/main.c.obj] Error 1
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/app.dir/all' failed
make[1]: *** [CMakeFiles/app.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/************************************************************************/

This error message also shown up when using file system, when have head file fs.h included (#include <fs.h> ).

miyatsu, 2017-12-12:
@SebastianBoe

miyatsu, 2017-12-12:
In hello world sample, do following steps, get compile error.

Step1: vim prj.conf, add:
CONFIG_FLASH=y
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_NFFS=y

Step2: vim src/main.c, add:
#include <fs.h> above the main function.

Step3:
mkdir -p build/frdm_k64f && cd build/frdm_k64f
cmake -DBOARD=frdm_k64f ./ ../../
make -j4

Step4: Output with error:

/************************************************************************/
In file included from /home/username/project/zephyr/include/fs/fs_interface.h:47:0,
                 from /home/username/project/zephyr/include/fs.h:11,
                 from /home/username/project/zephyr/samples/hello_world/src/main.c:10:
/home/trylong/project/zephyr/include/fs/nffs_fs.h:11:23: fatal error: nffs/nffs.h: No such file or directory
 #include <nffs/nffs.h>
                       ^
compilation terminated.
CMakeFiles/app.dir/build.make:302: recipe for target 'CMakeFiles/app.dir/src/main.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/src/main.c.obj] Error 1
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/app.dir/all' failed
make[1]: *** [CMakeFiles/app.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/************************************************************************/

Same error when use FAT file system.

SebastianBoe, 2017-12-12:
You are discovering a known, but undocumented usability issue.

This is now tracked here: https://github.com/zephyrproject-rtos/zephyr/issues/5351

Look at how other samples are using mbedTLS and fs. They are doing

target_link_libraries(app mbedTLS)

PS: I didn't investigate the FS issue but I assume it is the same as mbedtls.

closed 2017-12-12 by SebastianBoe