zephyrproject-rtos/zephyr #5351: Some libraries should automatically be linked with 'app'

SebastianBoe, 2017-12-12:
It is difficult to understand for which SW components an application's CMakeLists.txt will need to do

target_link_libraries(app mbedTLS)

and for which it doesn't.

SebastianBoe, 2017-12-12:
One possible solution is to divide the zephyr libraries into two types.

One type that is automatically linked with app, by boilerplate.cmake.
And another that is not.

Perhaps it could be kconfigured whether app should be linked with or not, with sensible
defaults per-library.

In any case, this seems like a solvable problem that is making the build system difficult to use
for application developers. So this should have high priority.

miyatsu, 2017-12-12:
I've check the project in ${ZEPHYR_BASE}/tests/subsys/fs/nffs_fs_api/

The test code seems no such problems.

Do you mean if I add this line https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/subsys/fs/nffs_fs_api/CMakeLists.txt#L38 in my app CMakeLists.txt, the compile error will disappear?

SebastianBoe, 2017-12-12:
Sorry, I only looked at the mbedTLS case. I'll have a look at the NFFS case as well ...

miyatsu, 2017-12-12:
I already test it when I say it, it works.
I've been looking for the solution for at lease 2 hours... until you shown up. You helped me a lot, thanks..

closed 2018-01-03 by nashif

koffes, 2018-05-03:
I also ran into the same issue when adding FAT support. 
By including either:
`zephyr_library_link_libraries(subsys__fs)` or `target_link_libraries(app subsys__fs)` I was able to continue. However, this seems to add some tests to the build and I get multiple definitions of main(). Any tips?

`zephyr/tests/ztest/libtests__ztest.a(ztest.c.obj): In function `main':
zephyr/tests/ztest/src/ztest.c:274: multiple definition of `main'
libapp.a(main.c.obj): ..build_folder/../main.c:101: first defined here`

SebastianBoe, 2018-05-03:
@koffes : To be clear, are you trying to build an application or a test?

If you are trying to build an application I would check if CONFIG_ZTEST was set, and if so find out why and/or turn it off.

koffes, 2018-05-03:
Thank you. You were right, a CONFIG_ZTEST=y was part of the configuration.

PR #5498 (2018-01-03): cmake: Introduce Zephyr interface libraries
Introduce Zephyr interface libraries.

It is difficult to understand for which libraries an application's CMakeLists.txt will need to do
`target_link_libraries(app <some_lib>)`
`target_include_directories(app <path_to_some_other_libs_include_dir>)`
and for which it doesn't.

Intuitively, some SW components, when KConfig-enabled,
are expected to be in the 'app's include path.

This fixes #5351.

The user scenario for enabling SW components that have header files that may be 
included by app now looks like this:

1. Find and enable the component in the [... cut]
commit 3c8331d3 (2018-01-03): cmake: Introduce Zephyr interface libraries

Introduce Zephyr interface libraries to fix #5351.

From the documentation:

A Zephyr interface library is a thin wrapper over a CMake INTERFACE
library. The most important responsibility of this abstraction is to
ensure that when a user KConfig-enables a library then the header
files of this library will be accessible to the 'app' library.

This is done because when a user uses Kconfig to enable a library he
expects to be able to include it's header files and call it's
functions out-of-the box.

A Zephyr interface library should be used when there  [... cut]