zephyrproject-rtos/zephyr #5355: qemu_x86/qemu_x86_nommu hangs on big executable files

pfalcon, 2017-12-12:
When executing "make run" with BOARD=qemu_x86/qemu_x86_nommu on the following file:

~~~
   text	   data	    bss	    dec	    hex	filename
 198386	 779936	  41768	1020090	  f90ba	zephyr.elf
~~~

QEMU justs hangs, not providing any ouput.

pfalcon, 2017-12-12:
That particular file contains a testsuite (tests to run, reference output to match against).

andrewboie, 2017-12-12:
What steps have you taken to debug this?
How do you know this is a function of file size, or related to the MMU?

andrewboie, 2017-12-12:
This bug report is severely insufficient.
Please do more diligence and gather proper data for a developer to reproduce the problem with their own builds. Sending just an ELF file isn't acceptable.

pfalcon, 2017-12-12:
Please have patience with community bugreports. Thank you.

closed 2017-12-12 by andrewboie