zephyrproject-rtos/zephyr #5358: CII Badge: Repeatable build process

agross-linaro, 2017-12-12:
The project MUST be able to repeat the process of generating information from source files and get exactly the same bit-for-bit result. If no building occurs (e.g., scripting languages where the source code is used directly instead of being compiled), select "not applicable"

Need to verify this is the case.

kestewart, 2017-12-13:
Have followed up with CII expert about the question of timestamp, and was pointed to: 
https://reproducible-builds.org/docs/timestamps/

Following up from that article I asked  if we run https://packages.debian.org/sid/strip-nondeterminism on Zephyr and prove that two builds yield the same result we'll be good for this?
Answer was: 
   You'll have to *test* that, but I expect the answer would be "yes".
   Even if it isn't, typically you don't have to do *much* else.
   Sometimes you have to force some kind of ordering (e.g., insert a "sort" somewhere).

@nashif - will this be hard to incorporate into the regular processes?

nashif, 2018-02-24:
@kestewart 

We can verify reproducibility by disabling timestamps, in general we do not have any timestamps when we build, we add those for development builds only and should not be enabled for release builds. I think the only thing we need is a testcase that verifies this for every commit and on a regular basis on master.

I just verified this and we do have repeatable builds.

This should be closed as "Met"

agross-linaro, 2019-02-20:
This appears to be met via testing.  We are documenting this via:
https://github.com/zephyrproject-rtos/zephyr/wiki/Repeatable-Builds-by-Architecture

closed 2019-02-20 by agross-linaro