zephyrproject-rtos/zephyr #5360: CII Badge: Use of deprecated functions/APIs

agross-linaro, 2017-12-12:
The project SHOULD avoid using deprecated or obsolete functions and APIs where FLOSS alternatives are available in the set of technology it uses (its "technology stack") and to a supermajority of the users the project supports (so that users have ready access to the alternative).

This is a little generic of a requirement.  We'd need to do a scan to check for deprecated/obsolete functions that have FLOSS alternatives.

nashif, 2018-02-24:
I am not aware of any APIs that we used that can be seen as deprecated, would mark this as Met and if something deprecated is discovered, mark it as a bug.

agross-linaro, 2018-03-07:
Agreed.

agross-linaro, 2019-02-20:
The project is self-contained and does use minimal external APIS, mostly from libc. We verify usage of those APIs on constant basis and remove any obsolete usage when found or reported.

closed 2019-02-20 by agross-linaro