zephyrproject-rtos/zephyr #5365: TC_PASS should be TC_FAIL

pdxjohnny, 2017-12-12:
https://github.com/zephyrproject-rtos/zephyr/blob/96efdbbf3386ef3d499325c5d4b42bf082e2280c/tests/net/dhcpv4/src/main.c#L160

lpereira, 2017-12-12:
I'm not sending this a PR just yet (it needs to be converted to a ZTEST; I used a lot of my time to fix the test), but this branch here [contains a slightly improved test](https://github.com/lpereira/zephyr/commit/d19a2a8064401abb3c78bdd5733da5f778f9b416).  It's the basic thing, but the DHCP Offer packet has been decoded (and there are comments explaining each field) and all addresses and timeouts are checked against the expected values.  This should help your testing efforts.

lpereira, 2017-12-12:
@pswarnak, could you please take a look at this one?

pswarnak, 2017-12-22:
Hi @lpereira,
FYI: QA team member @tarunkum is looking into this issue.

pswarnak, 2018-01-05:
Hi @lpereira,
@tarunkum has modified your [code ](https://github.com/lpereira/zephyr/commit/d19a2a8064401abb3c78bdd5733da5f778f9b416)for converting it to ZTEST.
Can you please push your changes, so he can update the modified on top of it.

lpereira, 2018-01-18:
@tarunkum can push his changes to his repository and open a PR, using the commit from my repository.

tarunkum, 2018-01-24:
Hi @lpereira,

I tried to test my changes on your commit, when I compile the code getting below error. 

../src/main.c:9:22: fatal error: sys/mman.h: No such file or directory
 #include <sys/mman.h>

To solve this I have installed sudo dnf install glibc-devel.i686
But after installing I can see the header files in /usr/bin/sys folder But still getting the same error. 

Please let me know how can I solve this issue.

lpereira, 2018-01-24:
@tarunkum Are you able to build a simple "hello, world" program on your Linux system? Something like:

```C
#include <stdio.h>

int main(void)
{
     printf("Hello, World!\n");
}
```

If you can, then you have successfully installed the system C compiler, linker, and C library include files.  If not, please consult the documentation for your Linux distribution to know which packages to install.

(BTW, I find it odd that the header files are installed in `/usr/bin/sys`; they should be, by default, in `/usr/include/sys`.)

tarunkum, 2018-01-25:
@lpereira I am able to build a sample "Hello, world" program on Linux system. 

As well as able to compile the below example also.

#include <stdio.h>

int main(void)
{
     printf("Hello, World!\n");
}

You can build it with a command like:
gcc -m32 -o hello hello.c

Getting issue only if I compile the dhcp test.

Error Statement :- 
/src/main.c.obj   -c ../src/main.c
../src/main.c:33:22: fatal error: sys/mman.h: No such file or directory
 #include <sys/mman.h>

This error is not coming in previous codes.

lpereira, 2018-01-25:
Do you have the package with kernel headers? (IIRC it's called `kernel-headers` in Fedora.)

tarunkum, 2018-01-29:
@lpereira I have checked in Fedora & get to know these below packages are installed on my system.

Package kernel-devel-4.13.16-100.fc25.x86_64
Package kernel-headers-4.13.16-100.fc25.x86_64

jukkar, 2019-05-22:
DHCPv4 test is currently using ztest -> closing.

closed 2019-05-22 by jukkar