zephyrproject-rtos/zephyr #5374: merge_config.sh can behave differently from merge_config.py

SebastianBoe, 2017-12-13:
Steps to reproduce:

1. Change the buildscripts to use merge_config.py.
2. ./scripts/sanitycheck -Tsamples/drivers/led_ws2812/

The sample fails to build with merge_config.py.

nashif, 2017-12-13:
yep, the difference in generated .config:
```patch
--- /tmp/config_sh	2017-12-13 13:38:35.169084378 -0500
+++ /tmp/config_py	2017-12-13 13:38:15.579243008 -0500
@@ -330,7 +330,7 @@
 # CONFIG_GPIO_STM32_PORTJ is not set
 # CONFIG_GPIO_STM32_PORTK is not set
 # CONFIG_SHARED_IRQ is not set
-# CONFIG_SPI_LEGACY_API is not set
+CONFIG_SPI_LEGACY_API=y
 CONFIG_SPI=y
 CONFIG_SPI_STM32=y
 CONFIG_SPI_STM32_INTERRUPT=y
@@ -401,22 +401,6 @@
 CONFIG_SYS_LOG_LED_STRIP_LEVEL=0
 CONFIG_LED_STRIP_INIT_PRIORITY=90
 # CONFIG_LED_STRIP_RGB_SCRATCH is not set
-# CONFIG_LPD880X_STRIP is not set
-CONFIG_WS2812_STRIP=y
-CONFIG_WS2812_STRIP_NAME="ws2812_strip"
-CONFIG_WS2812_STRIP_SPI_DEV_NAME="ws2812_spi"
-CONFIG_WS2812_STRIP_SPI_BAUD_RATE=5250000
-CONFIG_WS2812_STRIP_ONE_FRAME=0x7c
-CONFIG_WS2812_STRIP_ZERO_FRAME=0x60
-
-#
-# The following options determine channel data order on the wire.
-#
-CONFIG_WS2812_RED_ORDER=1
-CONFIG_WS2812_GRN_ORDER=0
-CONFIG_WS2812_BLU_ORDER=2
-CONFIG_WS2812_HAS_WHITE_CHANNEL=y
-CONFIG_WS2812_WHT_ORDER=3

 #
 # Build and Link Features
```

nashif, 2017-12-17:
The culprit is 

```patch
-# CONFIG_SPI_LEGACY_API is not set
+CONFIG_SPI_LEGACY_API=y
```

prj.conf has 

```
# CONFIG_SPI_LEGACY_API is not set
```

carlescufi, 2018-01-08:
@nashif @SebastianBoe can we close this as per https://github.com/zephyrproject-rtos/zephyr/pull/5569 ?

SebastianBoe, 2018-01-08:
@carlescufi : What you can do is put "fixes #5374" in the PR description or a commit message and this will be automatically closed when the RFC is merged.

https://help.github.com/articles/closing-issues-using-keywords/

PR #5569 (2018-01-12): scripts: kconfig: Replace conf with Python
fixes #5374
fixes #5419
closed 2018-01-12 by nashif