zephyrproject-rtos/zephyr #5378: Networking stack buffer management update

laperie, 2017-12-13:
This is an umbrella issue for the set of architectural changes and optimizations discussed during Zephyr Networking Forum, 12th of December 2017.  The forum materials are [under this link](https://github.com/zephyrproject-rtos/zephyr/files/1555201/Zephyr.Networking.Forum.December.2017.materials.pdf)

A key dependency to it is the new design of networking buffers, proposed by @jhedberg  in #3283 

In terms of implementation, we have identified the following breakdown: 

[TASK1] Introducing compound buffer to the net stack
      [ST1] Introduce the buffers to core stack
      [ST2] Validate 6LoWPAN and TCP scenarios
      [ST3] Prototype use of external buffers
 
[TASK2] Socket API update 

[TASK3] IP stack high-level API update
     [ST1] API update 
     [ST2] Update protocols 

[TASK4] IP offloading API update

andrewboie, 2017-12-15:
One thing I am curious about -- are we planning on introducing some method to have the network stack use a user-suppiled set of buffers or allocation function?
I think we are going to want to provide a way for User Mode threads to work with the network stack, which I think would require that the buffers involved be in a section of memory writable to user mode. As yet nobody has looked into this in detail, but I hope we can take this into consideration if we are doing an API overhaul.

pfalcon, 2018-01-10:
Here's an email response I posted on 2017-12-13, after the presentation mentioned in the description, and before many folks went on the vacation:

----
Thanks for yesterday's presentation. One thing which caught my
attention yet during pre-reviewing of the slides was that "[with the new
buf management scheme,] It can even be user statically allocated buffer
too !". We had a stakeholder doc circulated on our (Linaro's) side
asking for support of lwIP-style "ROM" buffers. There was no talks of
that from our side, because that requirements wasn't pushed hard, and
it was clear that with the old scheme, there's no good way to support
it.

So, I'm personally very positive about the new proposal. But at the
same time, it's quite "revolutionary" indeed. And we should settle on
the organizational aspects of it, on stakeholder expectation and actual
expected deliverables from it. So, here're high-level questions about
the proposal:

First part of the presentation evolved around things like: "API
consistency: seeking for convergence" and "Why socket API is not
compatible with other APIs?" So:

1. How does the new buffer management proposal address "API
consistency" part?
2. How does it address "Why socket API is not compatible with other
APIs" part?

What are the expectations in that regard at all?

And a specific technical clarification:

3. Can the authors of the proposal answer in yes/no manner the following questions: a) Do we give up on idea to use fixed-size, non-contiguous "data fragments" to store a single network packet? b) If so, do we switch to a scheme that each network packet is represented by a single contiguous data buffer?

pfalcon, 2018-01-10:
While folks were away, I tried to find answers to the questions above, in particular:

> 3. Can the authors of the proposal answer in yes/no manner the following questions: a) Do we give up on idea to use fixed-size, non-contiguous "data fragments" to store a single network packet? b) [...]

Slide 14 in the presentation above (https://github.com/zephyrproject-rtos/zephyr/files/1555201/Zephyr.Networking.Forum.December.2017.materials.pdf) would answer that:

> You could have a net_pkt holding a list of 3 net_buf:
One for the L2 headers
One for the IP headers
One for the actual payload, i.e a reference to the user buffer

So, per this design, we'd still have data fragments, but they might be on different sizes. And according to the @jhedberg's branch, normal allocation would happen from k_mem_pool.

@jhedberg, @jukkar, @tbursztyka : Can you please confirm this?

If we speak variable-length allocation, that immediately leads to the concerns of memory allocation.

tbursztyka, 2018-01-10:
1 - consistency: solved by going towards socket based only API (the new buffer management is not solving this per se, it is enabling new technical possibilities which can be used to remove net_pkt/net_buf from user side).
2 - I thought this was obvious, as soon as it gets rid of net_pkt/net_buf in user API side, so it goes towards socket based API... it will be possible to have all high-level APIs made on top of a socket type of API.
3 - yes there will be fragments, you cannot have a flexible solution that prevent this. Payload being a ref on user buffer is only happening on specific case (blocking recv/send for instance), but that's the use-case mostly happening on tiny targets, with limited memory: so this will be a nice solution to avoid buf copies etc...

pfalcon, 2018-01-10:
@tbursztyka, thanks for the responses.

> 1 - consistency: solved by going towards socket based only API

> as soon as it gets rid of net_pkt/net_buf in user API side, so it goes towards socket based API

That's actually one of my concern. Let me formulate it like this: if there're talks about huge refactors, why don't we just switch to BSD Sockets API at all? I'm not sure if you mean just that, and I don't remember such a possibility to have been discussed at the 2017-12-12 meeting.

> 3 - yes there will be fragments

Ok, so we have fragments, and there will be fragments. But another "dumb" question - why don't we leave fragments fixed-size, as they're now, and just make "split fragement metadata from payload" (on which I'm +100). What calls to go to variable-sized fragments? And if we go that far, what is the response to memory fragments concerns?

pfl, 2018-01-11:
> That's actually one of my concern. Let me formulate it like this: if there're talks about huge refactors, why don't we just switch to BSD Sockets API at all? I'm not sure if you mean just that, and I don't remember such a possibility to have been discussed at the 2017-12-12 meeting.

While doing all the buffer updating, we are aware that there won't be time or manpower for one big patch to change it all in one go. Thus the parts need to be fixed one chunk at a time while the system has to be kept working as before. And users, whomever they are, will not be happy if their only choice of networking is suddenly replaced. Time will tell whether net_context() can be "switched off" at all - I suspect the current APIs will be assimilated into the rest of the update code.

> 3 - yes there will be fragments
>
> Ok, so we have fragments, and there will be fragments. But another "dumb" question - why don't we leave fragments fixed-size, as they're now, and just make "split [...]

There is BT LE that by itself fragments IP into packets with very small sizes, there is 802.15.4 that also has small packet sizes but relies on 6LoWPAN to do the fragmenting, and finally, the vast minority can carry 1280 bytes at a time, but that is our least concern. So there will be sets of similarly sized fragments, but with different sizes depending on technology. Then again, upper level protocols and insert and append operations of variable sizes can be supported with the proposed changes, how the user data ends up mapped to interface level smaller packets needs some thinking and design (or then I missed something essential with the proposed solution). Same with memory fragmentation, so yes, there is work ahead for all of this.

tbursztyka, 2018-01-11:
> Let me formulate it like this: if there're talks about huge refactors, why don't we just switch to BSD Sockets API at all?

I actually said that during the presentation yes, that we may just do that move. It was not written in any slide though.

About work plan, we have to come with something. In the end I am not sure users will be unhappy to switch to a well known API such as socket.

tbursztyka, 2018-01-11:
Btw about fragmentation and stuff, I did not come up with all the details in the presentations because this would have been too technical and off subject. Obviously there will be a lot to think and do, but hopefully this have been solved in many other OSes, there is nothing to invent, just a lot of work.

pfalcon, 2018-01-11:
> I actually said that during the presentation yes, that we may just do that move. It was not written in any slide though.

Well, that's nice to hear. I definitely think it would help TCP-based applications. But there're 2 matters then:

1. If we want to switch to BSD Sockets API for implementation of application-level protocols, it's not blocked by anything, the API is already there. Yes, it has issues (yet to be discovered), and underlying stack needs improvements, bit these tasks are parallel, not lying on critical path of each other.
2. We'd still need an API below sockets, for extremely frugal configurations, and, possibly, UDP services.

nashif, 2018-01-11:
> If we want to switch to BSD Sockets API for implementation of application-level protocols, it's not blocked by anything, the API is already there. Yes, it has issues (yet to be discovered), and [...]

And thats what I said in the meeting, this is work that can happen now without having to wait for refactoring. But based on my current understanding, bsd sockets in zephyr are still missing a few critical APIs, correct me if I am wrong.

pfalcon, 2018-01-11:
> about fragmentation... Obviously there will be a lot to think and do, but hopefully this have been solved in many other OSes, there is nothing to invent, just a lot of work.

Yeah, there's nothing to invent, and I'd say, there's nothing to solve either. It's as simple as: as long as you allocate objects of different sizes, there will be fragmentation, which will lead to memory waste, or put it another way, system memory capacity will be less than ideal expected. Moreover, for some systems, like realtime ones, the worst-case scenario should be considered and accounted for. And apparently, Zephyr should be considered such as system.

I'm not sure what to quote as a full-fledged theoretical reference, but I usually find a summary provided by the Sqlite project to be good enough: https://www.sqlite.org/malloc.html#_mathematical_guarantees_against_memory_allocation_failures . It says that fragmentation is a function of the ration of (maximum allocated size) / (minimum allocated size). For example, if there're 2 allocation sizes - 128 and 256 bytes, ration is 2, and half of the memory can be wasted in the worst case. It's easy to picture such a scenario - that's when 128 byte blocks are interleaves with 128 bytes of free space. In such arrangement, allocation for 256 bytes fails, even though only half of memory is utilized.

The higher the ratio, the worse the worst-scenario. I don't think that we want to build Zephyr on such a foundation. Consequently, fixed-size chunks, as they're now, is the only robust and efficient solution. The only variation is the size of chunks, many IP stacks take that to be the maximum (reasonable) size of a packet.

pfalcon, 2018-01-11:
> But based on my current understanding, bsd sockets in zephyr are still missing a few critical APIs, correct me if I am wrong.

Please define "critical APIs". The BSD Sockets API implementation in Zephyr misses a lot (and at the same time, very lightweight), and nobody so far reported that they miss anything at all. Show me a detailed report, and we can work from there. In the meantime, there're 4 in-tree samples: https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/sockets , and one out-of-tree sample (MicroPython socket module) which work pretty well with the current implementation.

So, they can be a good start for someone else to work on more socket things. At which point we'll of course be getting problem reports and can work on them. I personally don't go for more socket stuff because I don't want it to look like one faction stubbornly clings to sockets while another stubbornly clings to net_app (so prefer to work on common underlying issues, while trying to collect arguments and cases for/against one or another solution).

nashif, 2018-01-11:
for example, do we have support for select()?

In a conversation I had with @GAnthony he listed of APIs missing he discussed with you at some point to support offloading.

pfalcon, 2018-01-11:
> for example, do we have support for select()?

No, but neither it's needed in the context of "what if we rewrite Zephyr's in-tree app protocol implementation in sockets?". select() is effectively a legacy, limited and inefficient function. Instead, we have poll(). If using inside Zephyr tree, we'd use it. Actually, if we'd use it, then the next step, we'd need to optimize it by blending POSIX poll() requirements into Zephyr's k_poll(). (Another route might be to implement epoll(), but that's not POSIX).

> In a conversation I had with @GAnthony he listed of APIs missing he discussed with you at some point to support offloading.

Yeah, and I'd expect him to submit a ticket for that, but he doesn't for some reason ;-). And I can't just theorize and submit a ticket myself that it would be nice to have, because again, it's superseded by poll(). So, until people show viable usecases for what it's needed for (and that arguably would be usecases of porting 3rd-party apps), then it's probably not needed.

carlescufi, 2018-01-11:
@glennrub @rlubos FYI

GAnthony, 2018-01-11:
> Yeah, and I'd expect him to submit a ticket for that, but he doesn't for some reason ;-)

Read the arguments for poll vs select, and agree poll should be sufficient.  select() would be nice to have, for porting apps/protocols which use that API, and since some offload devices export select() instead of poll() - but perhaps poll() can be built over select(). 
In any case, it doesn't seem that missing select() API will be blocking the port of protocols to socket APIs.

GAnthony, 2018-01-11:
> If we want to switch to BSD Sockets API for implementation of application-level protocols, it's not blocked by anything, the API is already there. 

Yes, before the holidays, as an exercise, I started converting the Zephyr MQTT protocol to sockets (from net_app), and linear buffers (from net_pkt).   
So far, I plan to use poll(), and report if I hit any major difficulties.

pfalcon, 2018-01-11:
> but perhaps poll() can be built over select()

It definitely can be, the question is overhead of implementation. Anyway, as I mention, feel free to submit a ticket for implementing select(), and we can plan for it.

pfalcon, 2018-01-17:
While we have a spoken meeting scheduled for Jan 23, it would be nice continue discussion here, so we can target some actual decisions for the meeting.

@jhedberg, @tbursztyka : Would be nice to hear your responses to the fragmentation concerns above. (I'd especially like to ping @jhedberg , as the owner of the only code available for review now ;-) ).

pfalcon, 2018-01-23:
Some further discussion in https://github.com/zephyrproject-rtos/zephyr/pull/5721

pfalcon, 2018-12-06:
@tbursztyka : Probably worth posting an update here.

jukkar, 2019-02-24:
This issue is quite old and mostly bitrotted. We have the code mostly changed/converted anyway, there are couple of PRs still pending. I suggest that this is closed.

jukkar, 2019-03-09:
Closing this as the net_pkt work is done.

closed 2019-03-09 by jukkar