zephyrproject-rtos/zephyr #5387: Many of the samples using mbedtls_ssl_conf_psk() dont check for error

mike-scott, 2017-12-14:
During a review it was noticed that several of the current samples don't check the return value of mbedtls_ssl_conf_psk().  This should be fixed as this function can fail for various reasons.

- samples/net/coaps_server/src/coaps_server.c:510:	ret = mbedtls_ssl_conf_psk(&conf, psk, strlen((char *)psk),
- samples/net/mbedtls_sslclient/src/mini_client.c:223:	mbedtls_ssl_conf_psk(&conf, psk, sizeof(psk),
- samples/net/echo_client/src/udp.c:96:	mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
- samples/net/echo_client/src/tcp.c:96:	mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
- samples/net/coaps_client/src/coaps_client.c:258:	ret = mbedtls_ssl_conf_psk(&conf, psk, strlen((char *)psk),
- samples/net/http_client/src/main.c:117:	mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
- samples/net/mqtt_publisher/src/main.c:88:	mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,`

galak, 2018-12-06:
Looks like we've moved forward on this, but still a few cases need to be addressed:

samples/net/echo_client/src/tcp.c:      mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
samples/net/echo_client/src/udp.c:      mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
samples/net/http_client/src/main.c:     mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,

jukkar, 2019-02-21:
> samples/net/echo_client/src/tcp.c: mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
> samples/net/echo_client/src/udp.c: mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,
> samples/net/http_client/src/main.c: mbedtls_ssl_conf_psk(&ctx->tls.mbedtls.conf,

These samples are already removed so closing this.

closed 2019-02-21 by jukkar