zephyrproject-rtos/zephyr #5401: delta_ticks_from_prev become negative and waiting tasks never scheduled

greenhandholman, 2017-12-14:
After a suspend period, if you wake up late and announce to the kernel a greater number of ticks  than the requested suspend period, timeouts are never processed in _handling_timeouts function because delta_ticks_from_prev become negative. 

To resolved this issue, i have make the following modification with sucess. Is anyone seeing a problem with this change?

kernel/sys_clock.c
@@ -240,7 +240,10 @@ static inline void handle_timeouts(s32_t ticks)
return;
}
- head->delta_ticks_from_prev -= ticks;
+ if (head->delta_ticks_from_prev > ticks)
+ head->delta_ticks_from_prev = 0;
+ else
+ head->delta_ticks_from_prev -= ticks;
/*
* Dequeue all expired timeouts from _timeout_q, relieving irq lock

greenhandholman, 2017-12-15:
After some tests and reflections, I noticed that we had to apply the delay to all waiting times. To do this I propose the following modifications.

kernel/sys_clock.c
```diff
@@ -240,8 +240,6 @@ static inline void handle_timeouts(s32_t ticks)
 		return;
 	}
 
-	head->delta_ticks_from_prev -= ticks;
-
 	/*
 	 * Dequeue all expired timeouts from _timeout_q, relieving irq lock
 	 * pressure between each of them, allowing handling of higher priority
@@ -254,27 +252,37 @@ static inline void handle_timeouts(s32_t ticks)
 
 	_handling_timeouts = 1;
 
-	while (timeout && timeout->delta_ticks_from_prev == 0) {
+	while (timeout) {
+		if (timeout->delta_ticks_from_prev <= ticks) {
+			ticks = ticks - timeout->delta_ticks_from_prev;
+			timeout->delta_ticks_from_prev = 0;
+		}
+		else
+			timeout->delta_ticks_from_prev -= ticks;
 
-		sys_dlist_remove(next);
+		if (timeout->delta_ticks_from_prev == 0) {
+			sys_dlist_remove(next);
 
-		/*
-		 * Reverse the order that that were queued in the timeout_q:
-		 * timeouts expiring on the same ticks are queued in the
-		 * reverse order, time-wise, that they are added to shorten the
-		 * amount of time with interrupts locked while walking the
-		 * timeout_q. By reversing the order _again_ when building the
-		 * expired queue, they end up being processed in the same order
-		 * they were added, time-wise.
-		 */
-		sys_dlist_prepend(&expired, next);
+			/*
+			 * Reverse the order that that were queued in the timeout_q:
+			 * timeouts expiring on the same ticks are queued in the
+			 * reverse order, time-wise, that they are added to shorten the
+			 * amount of time with interrupts locked while walking the
+			 * timeout_q. By reversing the order _again_ when building the
+			 * expired queue, they end up being processed in the same order
+			 * they were added, time-wise.
+			 */
+			sys_dlist_prepend(&expired, next);
 
-		timeout->delta_ticks_from_prev = _EXPIRED;
+			timeout->delta_ticks_from_prev = _EXPIRED;
+		}
 
 		irq_unlock(key);
 		key = irq_lock();
 
 		next = sys_dlist_peek_head(&_timeout_q);
+		if (next == &timeout->node)
+			break;
 		timeout = (struct _timeout *)next;
 	}
 

 ```

andrewboie, 2017-12-15:
Can you send this as a PR so it can be reviewed?

PR #5424 (2018-01-30): kernel: Allow late processing of timeouts
This change proposes to handle the case where the handle_timeouts
function is called after a number of ticks greater than the first
timeout delta of the _timeout_q list. In the current implementation if
the case occurs, after subtracting the number of ticks the
delta_ticks_from_prev field becomes negative and the first timeout is
never processed. It is therefore necessary to treat this case and to
prevent delta_ticks_from_prev from becoming negative. Moreover, the lag
produced by the initial delay must also be applied to following timeouts
by browsing  [... cut]
commit 8375fb76 (2018-01-30): kernel: Allow late processing of timeouts

This change proposes to handle the case where the handle_timeouts
function is called after a number of ticks greater than the first
timeout delta of the _timeout_q list. In the current implementation if
the case occurs, after subtracting the number of ticks the
delta_ticks_from_prev field becomes negative and the first timeout is
never processed. It is therefore necessary to treat this case and to
prevent delta_ticks_from_prev from becoming negative. Moreover, the lag
produced by the initial delay must also be applied to following timeouts
by browsing [... cut]
closed 2018-01-30 by nashif