zephyrproject-rtos/zephyr #5406: UART1 on STM32F0 Series not working: errors in makro to enable clock

DWagenk, 2017-12-15:
STM32F0 Series has no real APB2 clock, but uses APB1 GRP2 instead.

UART initialization code in drivers/serial/uart_stm32.c for UART1 (UART6 to UART8 might be effected, too, but they are not present on the SoCs I have available) tries to initialize STM32_CLOCK_BUS_APB2 but should Initialize STM32_CLOCK_BUS_APB1_2 instead.
Same for the generated LL_APB2_GRP1_PERIPH_USART1 instead of LL_APB1_GRP2_PERIPH_USART1 that would be correct for F0 Series.

My temporary fix, generic enough, that it could work as a general fix, but still not beatuiful:
https://gist.github.com/DWagenk/6c37b20b3ffd8f92646d901ec25104ee#file-uart_stm32-c-diff

Anybody here that has a better fix? maybe just having a different #define for STM32_CLOCK_UART(...) on F0 that can "translate" **APB2** to **APB1, GRP2**?

erwango, 2017-12-15:
I wonder if we should finally keep this whole macro thing.
It's not the easiest to read, does not get better in time and might end up using more code than initial declarations.

DWagenk, 2017-12-20:
fix merged to arm branch. closing it now.

PR #5425 (2017-12-19): use the right clock busses for STM32F0 series UARTs
This fixes #5406.

removed the string-concatenation in clock init macro and give full clock bus names in UART init functions instead.
@erwango please review
commit 98461980 (2017-12-19): drivers: serial: use the right clock busses for STM32F0 series UARTs

STM32F0 Series uses different clock busses for UARTs than other
STM32FX Series.

fixes #5406
commit 32ee8dd0 (2018-01-11): drivers: serial: use the right clock busses for STM32F0 series UARTs

STM32F0 Series uses different clock busses for UARTs than other
STM32FX Series.

fixes #5406
commit 1f66015f (2018-01-12): drivers: serial: use the right clock busses for STM32F0 series UARTs

STM32F0 Series uses different clock busses for UARTs than other
STM32FX Series.

fixes #5406
closed 2017-12-20 by DWagenk