zephyrproject-rtos/zephyr #5408: Improve docs & samples on device tree overlay

galak, 2017-12-15:
Work up a sample to show usage of device tree overlay and document the sample so a user / app developer knows how to utilize overlay

DWagenk, 2017-12-18:
thanks, your e-mail response already helped me a lot

> You should be able to use an device tree overlap file.  There’s examples in ./samples/bluetooth/hci_uart. The overlay file needs to be named BOARD_NAME.overlay.

mbolivar-nordic, 2020-04-23:
@galak I believe this is fixed now: https://docs.zephyrproject.org/latest/guides/dts/howtos.html#set-devicetree-overlays

Please reopen if you disagree. I don't think a whole sample is needed

closed 2020-04-23 by mbolivar-nordic

KenthJohan, 2022-08-23:
More whole overlay samples please. I have a hard time filling in the gaps of incomplete examples to make it compile and work.