zephyrproject-rtos/zephyr #5418: Provide a python based replaced for gperf

nashif, 2017-12-17:
To support native builds on windows, look into providing a python based replaced for gcov

carlescufi, 2018-01-08:
@lpereira now that we're getting closer to Kconfig and DTC, can you give us an update on this?

carlescufi, 2018-01-08:
One more thing, `gperf` is actually portable and can be easily built on Windows. This obviously requires generating the binary, but it might be easier than writing a Python version. See: http://git.savannah.gnu.org/cgit/gperf.git/tree/README.windows

lpereira, 2018-01-08:
I have some code to generate a perfect hash table with Python, but the generated hash function uses a MOD operator, which is not a good idea.  (Gperf generates a function that's essentially a few array lookups, which is much better suited for microcontrollers.)  I'll take a closer look at this issue this week.

In any case, there are a few things that can be improved WRT gperf, at least for this application, that I want to explore with the Python gperf equivalent:

1. All kernel pointers start at a known, fixed address.  This offset can be subtracted from all pointers added to the hash table; this will generate smaller numbers, requiring less bits to represent them.
2. Those pointers are aligned at 4-byte boundaries; they can be divided by 4, reducing the number of bits even further.

Some quick back of the envelope calculations got me some really nice numbers: storing the offset divided by 4 will cut the memory required to store these pointers for lookup by half in the average case.

I've also experimented with other ways to store the same information.  For instance, by using the offset/4 as an index to a bitmap, it's possible to know if a pointer is valid or not; if it's valid, offset/4 can be used to index an array with type information as it is done today.  This might be an even better approach than a perfect hash table.

lpereira, 2018-01-08:
The last idea works very well -- here's some PoC in a [gist](https://gist.github.com/lpereira/787cd97aa0fefaa93af81c6ae91f0a07). Just need to use that to generate a C file.

carlescufi, 2018-01-08:
@lpereira I can easily create a chocolatey package for gperf on Windows, since gperf builds fine on it. I’m fine either way, but just letting you know since then after for the user it will be as simple as choco install gperf

carlescufi, 2018-01-09:
@lpereira I just managed to build `gperf` with Visual Studio's compiler, so it doesn't get any more native than this. I will try to create a chocolatey package now and report back.

carlescufi, 2018-01-09:
I've now submitted a package containing both 32 and 64-bit executables to chocolatey: https://chocolatey.org/packages/gperf/3.1
@lpereira feel free to push your replacement, but we now have a native Windows version of `gperf`

lpereira, 2018-01-10:
Thanks, @carlescufi!  This will give me more time to work on the Python replacement (the gperf generation is quite rube-golgbergy, and I'd like to clean it up), as I'm currently swamped with other things.

carlescufi, 2018-01-23:
Closing this because we have a fully native Windows executable.

closed 2018-01-23 by carlescufi