zephyrproject-rtos/zephyr #5428: can not build for esp32

watershade, 2017-12-18:
After I set ESP_IDF_PATH and others things, I just can not compile it .
I get message:
`  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD esp32
Zephyr version: 1.10.99
Using /home/galileo/Workspace/zephyr/boards/xtensa/esp32/esp32_defconfig as base
Merging /home/galileo/Workspace/zephyr/samples/hello_world/prj.conf
Merging /home/galileo/Workspace/zephyr/tests/include/test.config
#
# configuration written to /home/galileo/Workspace/zephyr/samples/hello_world/build/esp32/zephyr/.config
#
-- Generating zephyr/include/generated/autoconf.h
CMake Error at /home/galileo/Workspace/zephyr/cmake/extensions.cmake:878 (message):
  No such file or directory: CMAKE_READELF:
  '/home/galileo/Workspace/esp/ESPRESSIF_TOOLCHAIN_PATH/bin/xtensa-esp32-elf-readelf'
Call Stack (most recent call first):
  /home/galileo/Workspace/zephyr/cmake/toolchain-gcc.cmake:14 (assert_exists)
  /home/galileo/Workspace/zephyr/cmake/toolchain.cmake:16 (include)
  /home/galileo/Workspace/zephyr/cmake/app/boilerplate.cmake:201 (include)
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!
`

lpereira, 2017-12-18:
I just built the hello world app for ESP32 without any issues here.  What are the values of some of the environment variables?  You can check with:

```
echo $ESPRESSIF_TOOLCHAIN_PATH
echo $ESP_IDF_PATH
echo $ZEPHYR_GCC_VARIANT
```

watershade, 2017-12-22:
Thanks, I will try again

closed 2017-12-22 by watershade