zephyrproject-rtos/zephyr #5450: Enable -Wshadow by default

nashif, 2017-12-20:
Warn whenever a local variable or type declaration shadows another variable, parameter, type, class member.

dleach02, 2021-07-09:
@nashif I see that this -Wshadow can be enabled with the -DW=<1|2|3>. It generates a lot of warnings though.

dcpleung, 2023-08-31:
https://github.com/zephyrproject-rtos/zephyr/pull/61523 partially enables `-Wshadow` on the main Zephyr repo for code under `arch/`, `boards/`, `drivers/`, `kernel/`, `lib/`, `soc/`, and `subsys/` (as describe in commit e38fc6de8ace95299595ff7963a02f9dfa712cd2). For `samples/`, and `tests/`, they require enabling `-Wshadow` for the whole tree as there does not seem to be an easy way to achieve this without modifying all `CMakeLists.txt` under these directories. At this point, we need to fix shadow variables on various modules before we can enable `-Wshadow` for whole tree.

nordicjm, 2026-09-30:
almost time for the 10 year reunion of this issue, seems the checks are enabled in most areas just not on module code @nashif any easy way to run CI in a PR that would builds a number of code using modules so we can see the impact?