zephyrproject-rtos/zephyr #5460: scripts: extract_dts_includes: check yaml dts binding property "category: "

erwango, 2017-12-21:
yaml dt-binding files document a property category: <optional | required>
Aim is to document that a dts node property should be present or not.

This property is currently not checked, which makes it bogus.

Check the property when dts file is parsed, and trigger an error when a property node tagged "required" is not present.

mbolivar, 2017-12-21:
Thanks for raising this! I agree this issue seems valid. @agross-linaro, any comments?

erwango, 2017-12-21:
@galak: We recently discussed about "improve error checking of yaml in the future".
I think this issue belongs to this topic.
If you already have pre-defined requirements, let's put them here, so this issue could be solved in the good direction from start.

erwango, 2019-02-13:
@ulfalizer , maybe you can take this one, while you're cleaning out the script?

erwango, 2019-02-15:
@ulfalizer , I hope you don't mind I assign this to you since you are in the process of heavily reworking the script.

pabigot, 2019-07-06:
@galak @ulfalizer Is this being actively worked in the scripting overhaul?  Is that overhaul being tracked somewhere?  The lack of robust device tree validation and presence of quirky behavior is very disheartening.

galak, 2019-07-30:
@erwango is this still relevant with the new dts generation scripts?

erwango, 2019-07-31:
Tested and working indeed.
One question though, I tested setting `category: required` to a `type: boolean`. If prop is not present, no message is generated. @ulfalizer is that an intended behavior?

galak, 2019-07-31:
> One question though, I tested setting `category: required` to a `type: boolean`. If prop is not present, no message is generated. @ulfalizer is that an intended behavior?

If the prop is not in the dts?  Than this is correct behavior as a 'boolean' is allowed to not exist in the .dts

pabigot, 2019-07-31:
Note that in this special case a define is generated for `DT_PROPERTY=0`, unlike other property types where the absence is indicated by no generated define.  This is a little inconsistent, but can be dealt with.

erwango, 2019-08-01:
Fine with me then. Let me close this point.

closed 2019-08-01 by erwango