zephyrproject-rtos/zephyr #5467: NFFS file system does no build when newlib libc is used

vanwinkeljan, 2017-12-21:
The NFFS file system does not build when newlib libc is used due to type conflicts in NFFS os specific function  implementations. This is caused due to the fact that NFFS uses uint32_t (defined as unsigned long in case of newlib) in the prototypes but zephyr uses u32_t (defined as unsigned int) in the implementations.

lpereira, 2017-12-22:
Shouldn't the fix in #5468 be the other way around, fixing the declaration of `u32_t` to match `uint32_t`?  (I never bought the argument for these short types, but... pick your battles, I guess.)

vanwinkeljan, 2017-12-23:
@lpereira  Actually that was my first fix where I update zephyr/types.h to typedef uint32_t to u32_t but I thought this was a bit to intrusive.

PR #5468 (2018-01-02): fs: nffs: Corrected types in OS specific functions
Corrected types in OS specific functions to be inline with NFFS
declarations.

Fixes #5467
commit 36e2efaf (2018-01-02): fs: nffs: Corrected types in OS specific functions

Corrected types in OS specific functions to be inline with NFFS
declarations.

Fixes #5467
closed 2018-01-02 by nashif