zephyrproject-rtos/zephyr #5471: cmake errors at -B containing @

inakypg, 2017-12-22:
Hello

When I try to issue a build directory that contains an @ sign, cmake complains; this is quite a problem because autobuilders such as make will issue @ as a workspace path for when running multiple jobs in parallel -- definitely breaks compatibility with previous behaviour.

```
$ mkdir build@4
$ cmake -DZEPHYR_GCC_VARIANT=zephyr -DBOARD=arduino_101 -H$ZEPHYR_BASE/samples/net/http_server/ -B$PWD/build@4/
...
-- Performing Test check_Wl___print_memory_usage
-- Performing Test check_Wl___print_memory_usage - Success
CMake Error at /home/inaky/z/kernel.git/cmake/extensions.cmake:310 (add_custom_target):
  The target name
  "_home_inaky_z_build@4_zephyr_include_generated_echo-apps-cert_der_inc" is
  reserved or not valid for certain CMake features, such as generator
  expressions, and may result in undefined behavior.
Call Stack (most recent call first):
  CMakeLists.txt:16 (generate_inc_file_for_target)

CMake Error at /home/inaky/z/kernel.git/cmake/extensions.cmake:310 (add_custom_target):
  The target name
  "_home_inaky_z_build@4_zephyr_include_generated_echo-apps-key_der_inc" is
  reserved or not valid for certain CMake features, such as generator
  expressions, and may result in undefined behavior.
Call Stack (most recent call first):
  CMakeLists.txt:16 (generate_inc_file_for_target)

CMake Error at /home/inaky/z/kernel.git/cmake/extensions.cmake:310 (add_custom_target):
  The target name
  "_home_inaky_z_build@4_zephyr_include_generated_index_html_inc" is reserved
  or not valid for certain CMake features, such as generator expressions, and
  may result in undefined behavior.
Call Stack (most recent call first):
  CMakeLists.txt:16 (generate_inc_file_for_target)

-- Configuring incomplete, errors occurred!
See also "/home/inaky/z/build@4/CMakeFiles/CMakeOutput.log".
See also "/home/inaky/z/build@4/CMakeFiles/CMakeError.log".

```

nashif, 2017-12-22:
who adds the "@"? Has this worked before with another version of cmake? which one?

inakypg, 2017-12-22:
jenkins is adding the @  -- when it defines a workspace, say XYZ/jobname, when it has to launch multiple jobs in the same slave, they are launched at XYZ/jobname@2, @3, etc..

This has worked like this for ever, however -- we noticed only on a few jobs because it only happens when there is more load and more jobs happen in the same slave. 

It did work with Kbuild

nashif, 2017-12-22:
fix should be simple actually:

```patch
diff --git a/cmake/extensions.cmake b/cmake/extensions.cmake
index a1a1219..0c20ce8 100644
--- a/cmake/extensions.cmake
+++ b/cmake/extensions.cmake
@@ -306,6 +306,7 @@ function(generate_inc_file_for_target

   string(REPLACE "/" "_" generated_target_name ${generated_target_name})
   string(REPLACE "." "_" generated_target_name ${generated_target_name})
+  string(REPLACE "@" "_" generated_target_name ${generated_target_name})

   add_custom_target(${generated_target_name} DEPENDS ${generated_file})
   add_dependencies(${target} ${generated_target_name})
```

PR #5472 (2017-12-23): cmake: replace @ in path with _
This fixes #5471
closed 2017-12-23 by nashif