zephyrproject-rtos/zephyr #5481: net: Another duplicate function to get pkt address

pfalcon, 2017-12-26:
Following https://github.com/zephyrproject-rtos/zephyr/issues/4421 , this is 3rd function with almost the same functionality (previous duplication case fixed). Well, this time, it has signature:

~~~
static void pkt_get_sockaddr(sa_family_t family, struct net_pkt *pkt,
                             struct sockaddr_ptr *addr)
~~~

i.e., it takes `struct sockaddr_ptr*`, whereas `net_pkt_get_dst_addr()` takes `struct sockaddr*`. Still, I guess it's worth looking into to avoid duplication.

pfalcon, 2017-12-26:
@askawu : Let me know if you're interested to look into this, based on the previous experience. If not, this is in my queue.

askawu, 2017-12-27:
@pfalcon , please assign this to me. :)

pfalcon, 2017-12-27:
@askawu : Done, thanks!

pfalcon, 2018-01-23:
@askawu : Ping, didn't hear from you for a while, is this still on your plate?

laperie, 2018-02-13:
it's actually a feature, not a bug ;)

pfalcon, 2018-02-13:
> it's actually a feature, not a bug ;)

Well, it's definitely not a "bug", but:

1. Objectively, a missed optimization opportunity. I didn't hear about code size comparison of Zephyr's IP stack with other IP stack, but given such cases, it's hard to shake off a hunch that we don't compare favorably.
2. It shows that we have some inconsistency/technical debt. Actually, somewhere around 1.10 I got an idea that we have a pretty vast and featureful stack, and no single party (except for hopefully @jukkar) knows it well enough, so there's a chance that a particular developer when adding new feature might reinvent/duplicate some functionality (even if partially).

I don't think that we're ready for full-fledged optimization (that's tracked in https://github.com/zephyrproject-rtos/zephyr/issues/3547), but IMHO, p.2 is a real concern, especially is we're talking about things like MISRA C (what's the point of MISRA C, which tells us how to write expressions and statements, if we have problems with how *code* is being written?).

Anyway, in the current situation, this is definitely prio:low.

pfalcon, 2018-02-13:
@nashif 

> you should try to assign bugs to one person only, otherwise everyone will be relying on the others

Per above, I try to assign @jukkar to any networking issue, to "keep him in loop". Not doing that would mean giving up an idea that "@jukkar for sure knows what happens in the IP stack, down to small ideas" ;-). He didn't raise any concerns, if there're suggestions for better workflow (e.g. just @mention), let's go over them.

To make the above more fair, I now assign ticket I might work on to myself too.

And @askawu volunteered to work on this, hence 3 assignees %). My guess he's on Chinese New Year vacation now, and I myself have enough a queue I think, so this isn't marked for 1.11.

jukkar, 2019-02-21:
This looks like a stale issue atm as there has been lot of changes in net_pkt API -> closing.

closed 2019-02-21 by jukkar