zephyrproject-rtos/zephyr #5482: net: RFC: Move TCP-related code from net_context.c to tcp.c

pfalcon, 2017-12-26:
Currently, we have following file sizes:

~~~
-rw-rw-r-- 1 pfalcon pfalcon 58127 Dec 26 21:11 net_context.c
-rw-rw-r-- 1 pfalcon pfalcon 32699 Dec 26 21:11 tcp.c
~~~

A *lot* of code in net_context.c is actually related to TCP handling, which would naturally belong to tcp.c instead. Given pretty big skew of the size towards net_context.c, I think one of the following should be done:

1. Move *some* TCP-related code to tcp.c.
2. Move *as much as possible* TCP-related code to tcp.c.

The risk of choice 2 might be that file sizes become 15K/75K. But if that's corresponds to the reality, why not? (One of the ideas behind this ticket is to be able to assess binary code size of various features too). Another "risk" is that net_context.c contains bunch of static TCP-related funcs, which would become public. Which is, again, fine too (actually, if move as-much-as-possible code, they will be static in tcp.c too).

pfalcon, 2017-12-26:
@jukkar, @tbursztyka, @lpereira, @pfl: Would appreciate your comments on this idea.

lpereira, 2017-12-26:
I would prefer option (2).  Two things matter here: readability and binary size.  Readability will improve, IMHO; it's kind of confusing to have protocol-specific things in what should have been a protocol-agnostic file.  Binary size shouldn't change, which is good; it might even improve slightly with more static functions available to the optimizer.

pfalcon, 2017-12-26:
@lpereira : Thanks!

pfl, 2018-01-02:
Yes, take option 2 and move it (all) to tcp.c. That's on a yellow sticker on @jukkar's desk and is waiting for someone to update the code.

pfalcon, 2018-01-25:
@pfl : Ah, good to know! Fairly speaking, this is a kind of task which is (much, subjectively) easier to do for a maintainer than for a contributor.

@jukkar : Any comments? Should be target this for 1.11 or definitely not?

pfl, 2018-02-20:
I'll fix this issue.

pfl, 2018-04-16:
This was merged some time ago.

closed 2018-04-16 by pfl