zephyrproject-rtos/zephyr #5483: Native port (POSIX) should accept command-line arguments

lpereira, 2017-12-26:
In order to help with fuzzing efforts, the native port should accept command-line arguments.

nashif, 2017-12-27:
good start is `--pid-file` :)

aescolar-ot, 2018-01-04:
One possible quick option is to base it on [argtable3](http://www.argtable.org/) which is 3 clause BSD licensed.
In that case I guess we would have a new external library like ext/lib/argparse/argtable3/ (?)
@nashif : Does this sound reasonable? Does it need to be addressed in the TSC?

aescolar-ot, 2018-01-04:
Another option is to use [argtable2](http://argtable.sourceforge.net/), which is provided as a package for quite a few distros (including Debian/Ubuntu), and is LGPL. But with it we would add one dependency more to Zephyr.

lpereira, 2018-01-04:
I don't think we need to use any library: just exposing `argc` and `argv` is fine.  I have a `[hackish patch](https://github.com/lpereira/zephyr/tree/dhcpv4-test)` that does this.

aescolar-ot, 2018-01-05:
Ok, it depends on what you want. If you just want to pass `argc` & `argv` to something else which will process them, or if you want to do the handling in native_posix board. (say check that `--pid-file` was there and dump the file).

aescolar-ot, 2018-01-05:
Maybe we should clarify what is that we try to achieve first.
My understanding of what different people wants (please correct me if I'm wrong):
* @nashif : The native_board should accept an argument like `[--pid-file <file_name>]`. When it gets it, it should dump to `file_name` the process id.
* In line with what Anas wants, I think it would be nice to add another argument `[--stop-at <time>]`. When it gets it, the program auto-stops after `time` seconds.
* @lpereira : An arbitrary set of arguments passed to the board main, should be left accessible for anybody who calls a function to pick them. Question: should that be all arguments? or do you want that the ones which have already been accepted by the top be removed from the list? Or would you like to get only what comes after some command line switch? (Say the user passes `[--testargs [arg1] [arg2] [...] ]` and you just get `[arg1] [arg2] [...]`

aescolar-ot, 2018-01-05:
Woudl something like this cover all these needs: https://github.com/aescolar-ot/zephyr/commit/ae5504361a7323c30aae8bdd47149d8823e9fd1c  [the code is horrible, it is just a demo]
Or what would be needed differently use wise?
```
$ zephyr/zephyr.exe --help
Usage:  [--help] [--stop-at=<time>] [--pid-file=filename] [arg]...

  --help                    display this help and exit
  --stop-at=<time>          In simulated seconds, when to stop automatically
  --pid-file=filename       Save the process id in this file
  arg                       Extra arguments for tests
```
And here an example of what you get when running it
```
$ zephyr/zephyr.exe melon --stop-at 1.0 banana tomato --pid-file somefile potato avocado & sleep 2 ; more somefile
[1] 1776
0: 'melon'
1: 'banana'
2: 'tomato'
3: 'potato'
4: 'avocado'
***** BOOTING ZEPHYR OS v1.10.99- - BUILD: Jan  4 2018 13:40:15 *****
Hello World! posix

Stopped after 1.000s
[1]+  Done                    zephyr/zephyr.exe melon --stop-at 1.0 banana tomato --pid-file somefile potato avocado
1776
```

aescolar-ot, 2018-01-09:
So as @nashif does not want his option by now, I'm just doing a very simple PR with what @lpereira wants.

PR #5612 (2018-01-11): native: native_posix accept command line arguments
The native_posix board will accept command line arguments
which the application / test may pick by calling
native_get_cmd_line_args()

Fixes #5483
closed 2018-01-11 by nashif