zephyrproject-rtos/zephyr #5485: tests: kernel/xip: CONFIG_BOOT_DELAY issue for qemu_riscv32

akhileshupadhyay, 2017-12-27:
This test case fails on introducing CONFIG_BOOT_DELAY.

Steps to reproduce:
1. cd Zephyr_Base/tests/kernel/xip
2. add CONFIG_BOOT_DELAY=10 to prj.conf.
3. mkdir build && cd build
4. cmake -DBOARD=qemu_riscv32 ../
5. make run

Below logs appear:
```
[QEMU] CPU: riscv32
▒qemu: hardware error: uart_write: bad write: addr=4 v=3fcc

CPU #0:
pc=0x00002694
 zero 00000000 ra   000026c8 sp   40001fd0 gp   00000000
 tp   00000000 t0   00000002 t1   0000000b t2   7fffffff
 s0   40002020 s1   400020cc a0   00003fe4 a1   40001ff4
 a2   00000002 a3   00000025 a4   00003fc4 a5   00003fcc
 a6   00010000 a7   0000000f s2   ffffffff s3   00000000
 s4   00000001 s5   00000000 s6   00000030 s7   00003fe4
 s8   00000000 s9   00000000 s10  40002104 s11  00000000
 t3   00000001 t4   00000004 t5   00000000 t6   00000000
 MSTATUS  00001800
 MIP      00000000
 MIE      00000000
FPR00: ft0 0000000000000000 ft1 0000000000000000 ft2 0000000000000000 ft3 0000000000000000
FPR04: ft4 0000000000000000 ft5 0000000000000000 ft6 0000000000000000 ft7 0000000000000000
FPR08: fs0 0000000000000000 fs1 0000000000000000 fa0 0000000000000000 fa1 0000000000000000
FPR12: fa2 0000000000000000 fa3 0000000000000000 fa4 0000000000000000 fa5 0000000000000000
FPR16: fa6 0000000000000000 fa7 0000000000000000 fs2 0000000000000000 fs3 0000000000000000
FPR20: fs4 0000000000000000 fs5 0000000000000000 fs6 0000000000000000 fs7 0000000000000000
FPR24: fs8 0000000000000000 fs9 0000000000000000 fs10 0000000000000000 fs11 0000000000000000
FPR28: ft8 0000000000000000 ft9 0000000000000000 ft10 0000000000000000 ft11 0000000000000000

```

galak, 2018-12-06:
Is this still relevant?

nategraff-sifive, 2018-12-06:
@galak I'm unable to reproduce this on the latest master (commit 3651e03eebbba5a8d281348d145ef8e03d8b65db) using SDK 0.9.5.

galak, 2018-12-06:
Ok, going to close this.  We can re-open if someone can reproduce.

closed 2018-12-06 by galak