zephyrproject-rtos/zephyr #5486: Bluetooth: Cannot connect to prevoiusly disconnected device when BT_PRIVACY is enabled

MariuszSkamra, 2017-12-27:
After successful disconnection, one cannot reconnect to the device.
Advertising packets from remote device are not received (remote advertises - advertisements are visible on another device) so that General Connection Establishment Procedure fails.

Configuration:
CONFIG_BT=y
CONFIG_BT_CENTRAL=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_SMP=y
CONFIG_BT_SIGNING=y
CONFIG_BT_GATT_CLIENT=y
CONFIG_BT_PRIVACY=y
CONFIG_BT_TINYCRYPT_ECC=y
[debug_privacy_enabled.log](https://github.com/zephyrproject-rtos/zephyr/files/1588886/iut-zephyr.log)
[traces_privacy_enabled.txt](https://github.com/zephyrproject-rtos/zephyr/files/1588883/traces_privacy_enabled.txt)

With CONFIG_BT_PRIVACY=n reconnection is successful.
[debug_privacy_disabled.log](https://github.com/zephyrproject-rtos/zephyr/files/1588856/iut-zephyr.log)
[traces_privacy_disabled.txt](https://github.com/zephyrproject-rtos/zephyr/files/1588866/traces_privacy_disabled.txt)

HW:
QEMU
Zephyr-based HCI BT controller running on nrf52840_pca10056
Head: 2e9064b7d646e97336f9aaa867487bd96514821b

Found during PTS testing.
Affected TC: GATT/CL/GAW/BV-02-C

MariuszSkamra, 2017-12-27:
I suspect controller issue

jhedberg, 2018-01-31:
Is this still reproducible? If I understand right it's the central side that's failing, right? We don't use whitelist-based scanning, so if the controller is not reporting anything to us this sounds like a potential controller issue. We do use controller-side resolving whenever possible, but that should cause the controller to generate at least some event for us, even if it doesn't have the right IRK.

MariuszSkamra, 2018-02-12:
Yes it is still reproducible. Zephyr is central in this test. 
I've tested this on QEMU with 3 different controllers:
1. Zephyr-based - FAIL
2. Intel Bluetooth 4.2 (0x08) - Revision 256 (0x0100) - FAIL
3. CSR PTS dongle - PASS

jhedberg, 2018-02-12:
I just tried to reproduce this and got a controller assert. @cvinayak could you take a look?

```
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                        #63 [hci1] 21.628034
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                   #64 [hci1] 21.628917
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection (0x08|0x000d) plen 25                                                     #65 [hci1] 21.631289
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: 50:F7:C0:4D:70:76 (Resolvable)
        Own address type: Random (0x01)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                     #66 [hci1] 21.633016
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: Vendor (0xff) plen 12                                                                            #67 [hci1] 21.826398
        aa 63 74 72 6c 2e 63 00 c5 05 00 00              .ctrl.c.....    
< HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0          
```

carlescufi, 2018-02-12:
@jhedberg https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/ctrl.c#L1477

Is this on nRF51?

jhedberg, 2018-02-12:
The controller assert seems to be reproducible every time. Steps to reproduce:

1. Apply the following patch to central_hr and peripheral_hr:

```
diff --git a/samples/bluetooth/central_hr/prj.conf b/samples/bluetooth/central_hr/prj.conf
index e937bf405d..c5721ad49b 100644
--- a/samples/bluetooth/central_hr/prj.conf
+++ b/samples/bluetooth/central_hr/prj.conf
@@ -3,3 +3,4 @@ CONFIG_BT_DEBUG_LOG=y
 CONFIG_BT_CENTRAL=y
 CONFIG_BT_SMP=y
 CONFIG_BT_GATT_CLIENT=y
+CONFIG_BT_PRIVACY=y
diff --git a/samples/bluetooth/peripheral_hr/prj.conf b/samples/bluetooth/peripheral_hr/prj.conf
index 08ebfd7e41..a448f8f81e 100644
--- a/samples/bluetooth/peripheral_hr/prj.conf
+++ b/samples/bluetooth/peripheral_hr/prj.conf
@@ -2,5 +2,6 @@ CONFIG_BT=y
 CONFIG_BT_DEBUG_LOG=y
 CONFIG_BT_SMP=y
 CONFIG_BT_PERIPHERAL=y
+CONFIG_BT_PRIVACY=y
 CONFIG_BT_DEVICE_NAME="Zephyr Heartrate Sensor"
 CONFIG_BT_DEVICE_APPEARANCE=833
```

2. Flash a combined build of peripheral_hr on some nRF board (I used the micro:bit)

3. Flash hci_uart on another nRF board (I used the nRF51 USB dongle), and attach it to BlueZ with the usual steps

4. Power on the board with peripheral_hr

5. Run central_hr in qemu using the hci_uart board as controller

jhedberg, 2018-02-12:
> @jhedberg https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/ctrl.c#L1477

> Is this on nRF51?

Yes, it is. An nRF51 USB dongle.

cvinayak, 2018-02-12:
@jhedberg and @MariuszSkamra you should not be able to select Controller Privacy for nRF51:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/Kconfig#L150

Could you check if it indeed has BT_CTLR_PRIVACY selected in your .config?

jhedberg, 2018-02-12:
Right, I completely forgot that the nRF51 can't really be used for testing controller privacy.

Anyway, as per your question @cvinayak CTLR_PRIVACY is *not* set for the hci_uart image I build (which asserted) and it's also not set for my micro:bit image. So I wouldn't have been able to do the full privacy test no matter what. Either way, I guess we want to fix the assert.

MariuszSkamra, 2018-02-12:
@cvinayak I'm reproducing this on nrf52840

cvinayak, 2018-02-12:
@MariuszSkamra do you see assert in nRF52840 too?

MariuszSkamra, 2018-02-12:
I enclose the sniffer traces from Ellisys analyzer for the failing and passing scenario.
I've tested with QEMU and Zephyr-based dongle (running on nrf52840).

[logs.zip](https://github.com/zephyrproject-rtos/zephyr/files/1716634/logs.zip)

jhedberg, 2018-02-13:
I can't reproduce the issue with central_hr against peripheral_hr with the nRF52840 with privacy enabled on both sides. It seems like you might have omitted a key step in the issue report: the devices need to be paired?

jhedberg, 2018-02-13:
I switched to using tests/bluetooth/shell on the central side and paired the devices. Still not able to reproduce the issue.

jhedberg, 2018-02-13:
I.e. I can reconnect just fine, as well as encrypt the new connection with the shared LTK.

cvinayak, 2018-02-13:
@MariuszSkamra and @jhedberg 

On analysis of the sniffer logs, there is advertisements during the test, for reconnection phase too for both cases; in the non-privacy log the address type used is random static address, and in the privacy log the address is random private resolvable address. these addresses do not change during the test.

In the privacy log, during the reconnection related advertising phase at packet number 3305, 3373, etc. I suspect possible reception of connect_ind from PTS which are not captured by sniffer. 

Please do a retest and capture for the failing private case, so that I can analyse the content of the reconnection connect_ind PDU.

MariuszSkamra, 2018-02-13:
I re-ran this several times and I failed to catch the CONNECT_IND PDU. Is it possible that PTS got LE Connection Complete event without sending this PDU?

cvinayak, 2018-02-13:
@MariuszSkamra Clear the nRF5 device has "seen" something that PTS seems to have sent which the sniffer does not see! The fact being the gap between advertising channels in the said packet numbers corresponding to 150 us (tIFS) + 352 us (CONNECT_IND) + ~70-120 us nRF52 CPU time to switch channels.

Without the contents of the CONNECT_IND or whatever is coming from PTS, its difficult to analyse further.

carlescufi, 2018-02-21:
@MariuszSkamra can you try to re-run the test to see if we can capture the elusive `CONNECT_IND` packet? This should be fixed before 1.11 is released.

MariuszSkamra, 2018-02-22:
@carlescufi I got CONNECT_IND event! You can check the parameters now
[connect_ind.zip](https://github.com/zephyrproject-rtos/zephyr/files/1748225/connect_ind.zip)

Test is failing for me again. That's odd. I'm afraid I tested vich patches with privacy disabled...

PR #6417 (2018-03-02): Bluetooth: Workaround privacy feature issue
This fixes intercompatibility issues with controllers supporting
privacy feature.
Core Spec requires to use network privacy mode as a default when
peer device provides its IRK during bonding when LL Privacy is used,
which is the case for Zephyr. We've seen devices including PTS
which exchanges it's IRK but is not aware about network privacy
mode. This results in Zephyr not able do be reconnect to such bonded
devices.
This workaround sets device privacy mode to be able to reconnect
to such devices.

Fixes #4989
Fixes #5486
commit 777f9c8c (2018-03-02): Bluetooth: Workaround privacy feature issue

This fixes intercompatibility issues with controllers supporting
privacy feature.
Core Spec requires to use network privacy mode as a default when
peer device provides its IRK during bonding when LL Privacy is used,
which is the case for Zephyr. We've seen devices including PTS
which exchanges it's IRK but is not aware about network privacy
mode. This results in Zephyr not able do be reconnect to such bonded
devices.
This workaround sets device privacy mode to be able to reconnect
to such devices.

Fixes #4989
Fixes #5486
closed 2018-03-02 by carlescufi