zephyrproject-rtos/zephyr #5501: Deprecate qemu.cmake in favour of a qemu runner backend

SebastianBoe, 2017-12-28:
Hi,

I suspect we want to eventually rename/alias 'flash' to 'run'. And treat qemu emulation as just
another runner backend. I believe this would make it easier for us to maintain the runner and qemu
infrastructure, and would make it easier for us to keep the user-interface for flashing and running
consistent.

A qemu backend would also allow us to move all the symlink/fifo/pipehandling out of qemu.cmake and
into a python runner backend.

It is not clear how difficult it would be to do this.

SebastianBoe, 2017-12-28:
@mbolivar : Did you already intend to move in this direction? I see you have added a dummy qemu.py backend already.

mbolivar, 2018-01-04:
@SebastianBoe Happy new year!

That was a copy of a dummy qemu.sh that was already there when I started working on the runner backends. I created a dummy Python script just so people would have a place to start if they want to continue this work. I have no plans to do it.

SebastianBoe, 2018-01-05:
Happy new year :)

Do you think it might make sense to incorporate running qemu into the runner infrastructure?

mbolivar, 2018-01-05:
I definitely think it makes sense. It's just another way to run the code, after all. I haven't tried the new "native" port, but I think that would make sense too -- keeping interfaces uniform is good usability.

mbolivar, 2019-05-16:
Upgrading the priority since I've received several questions about this recently.

nashif, 2021-05-28:
see #35366

nordicjm, 2024-11-19:
Seems like this was implemented years ago?

mbolivar, 2024-11-19:
> Seems like this was implemented years ago?

no; https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/west_commands/runners/qemu.py has always been a stub waiting for a hero to step up and fill it in