zephyrproject-rtos/zephyr #5546: (Stupid) questions about coverage reports

pfalcon, 2018-01-03:
Coverage reporting via https://codecov.io/gh/zephyrproject-rtos/zephyr was recently introduced, and coverage reports usually lead to a bunch of question. It's proposed to have this ticket to dump them.

pfalcon, 2018-01-03:
"Randomly" skipped lines.

Let's have a look at https://codecov.io/gh/zephyrproject-rtos/zephyr/src/master/misc/printk.c#L103 (we need line 103, in Firefox, it's obscured when opening that link, a cute bug in codecov.io UI). So, it's not marked as either hit or missed, which usually means that a line is not subject to code coverage (e.g. a declaration/definition, #ifdef'ed out, etc.). But that block is clearly the code, not under #ifdef, a compiler could not prove it's unreachable, etc.

At line 112, we clearly see a missing coverage. Then at line 116 undefined lines again, then at 124 missing coverage again. So, it's on and off between "how it should be" and "not clear why it's that way".

pfalcon, 2018-01-03:
@nashif : Do we build coverage with `-O0` ?

SebastianBoe, 2018-01-03:
I also have some stupid questions;

I get different coverage results in the github email notification and the Codecov github issue comment and find this confusing. It looks like Codecov edit's its answer, is this necessary?

pfalcon, 2018-01-03:
> It looks like Codecov edits it's answer, is this necessary?

Apparently it does, a tradeoff is getting "spammed" on each coverage updated: 
https://docs.codecov.io/docs/pull-request-comments#section-behavior

nashif, 2018-01-03:
> It looks like Codecov edits it's answer, is this necessary?

coverage reports come from different jobs and are merged into one report, hence the edits. IMO this is better than updating results with a new comment every time.

nashif, 2018-01-03:
> @nashif : Do we build coverage with -O0 ?

No

nashif, 2018-01-03:
btw, if you are looking for another output to compare with and something that you can reproduce locally, see:

http://docs.zephyrproject.org/coverage/misc/printk.c.gcov.html

This can be generated with sanitycheck --coverage

nashif, 2018-01-03:
and final note, there are a few bugs and lots of missing coverage on codecov.io right now, I am working with the provider to resolve them.

pfalcon, 2018-01-03:
> "Randomly" skipped lines.

Addressed by https://github.com/zephyrproject-rtos/zephyr/issues/5548

pfalcon, 2018-01-25:
I assumed coverage issues were settled, and the reports are clear enough for people. Closing.

mbolivar, 2018-01-30:
Reopening as there are unaddressed issues.

The code coverage reports don't make sense to me. Consider this pull request:

https://github.com/zephyrproject-rtos/zephyr/pull/5899

This is a change to the behavior of the J-Link flasher, which should have no effect on code coverage results in the following areas as reported in the PR:

```
tests/kernel/obj_tracing/src/philosopher.c | 88.88% <0%> (-11.12%) |  
-- | -- | --
subsys/net/ip/net_pkt.c | 56.88% <0%> (+0.56%) |  
subsys/net/ip/net_if.c | 62.56% <0%> (+0.81%) |  
tests/net/rpl/src/main.c | 87.5% <0%> (+2.71%) |  
include/net/net_pkt.h | 89.05% <0%> (+3.64%) |  
subsys/net/ip/rpl.c | 25.67% <0%> (+6.03%) |  
subsys/net/ip/rpl.h | 86% <0%> (+8%) |  
```

nashif, 2018-01-30:
@aescolar-ot There seems to be something off with how the native posix board is run, i.e. it is not quite deterministic and yields different results, probably related to timing? This is just a guess, need to look.

aescolar-ot, 2018-01-30:
@nashif I was actually thinking of asking you about it.. 
There is something weird indeed, and I don't think it has to do with the native port itself.
For example this PR https://github.com/zephyrproject-rtos/zephyr/pull/5862 or https://github.com/zephyrproject-rtos/zephyr/pull/5872 should not have caused any change, yet codecov claims there is coverage changes in the network side.. 
I could only explain it if you are running some sample app that is just killed due to a time out, (and therefore it depends on the load how far it gets).

The only part that should be a bit undeterministic and could be run on some cases and not others is the threads cleanup part inside posix_core (as there is a race between the hw models main thread exiting the whole process and the Zephyr threads cleanup after themselves) [This does not really matter as the kernel cleans up anyhow].

aescolar-ot, 2018-01-31:
@nashif I have just checked what happens in one of the offending cases ( tests/kernel/obj_tracing ), and at least in this one it seems quite clear where the problem lies:
sanitycheck will kill the test as soon as it sees the ``PROJECT EXECUTION *`` line, but in this case the affected line 
https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/kernel/obj_tracing/src/philosopher.c#L62

https://codecov.io/gh/zephyrproject-rtos/zephyr/compare/2151b864d4750e9b476cbc140b6c4baff59b7296...ada23b3cdaf9879cb09321590670486a15c3ef6d/src/tests/kernel/obj_tracing/src/philosopher.c
will be executed after the test case has passed.
Therefore sanitycheck sending the SIGTERM signal is racing the execution of that philosopher code.
In some cases it stops before that line is reached, in others not.
You can see here a trace with timestamps of this testcase, where I added an extra print next to that line:
https://pastebin.com/XjeHyvrQ
(note that the times are simulated times, so it all happens in a blink)

If you want consistency, one option is to add back the call to `posix_exit()` when the testcase ends (`TC_END`). That way you will have always exactly the same code execute (at least for tests that use ztest)

aescolar-ot, 2018-02-01:
A separate issue: Codecov seems to report 1 hit per line, even in lines which are obviously hit many times.

pfalcon, 2018-02-01:
> A separate issue: Codecov seems to report 1 hit per line, even in lines which are obviously hit many times.

Do you have a link for that? I believe I saw a hit count somewhere in its reports. So, do you say that this count shows 1, while it's logically provable (e.g. from context) that a particular line was hit more?

aescolar-ot, 2018-02-01:
>Do you have a link for that?
For example:
https://codecov.io/gh/zephyrproject-rtos/zephyr/src/master/arch/posix/core/posix_core.c
posix_swap() is called each time the kernel does a swap.. That will be quite a few times during the whole sanitycheck run.

aescolar-ot, 2018-02-05:
A separate thing to be aware:
If you do changes to code which is not run by sanitycheck in any of the native_posix tests or samples, codecov will not tell that the diff coverage is 0%.
Instead it will tell you it is `n/a` and that those files are not tracked.

nashif, 2018-02-11:
this is now resolved. thank you @aescolar-ot

closed 2018-02-11 by nashif